riscv/sail-riscv #215: Easy model build without dependencies that require sudo acces

martinberger, 2023-02-10:
In the discussion #212 on v0.15 planning, @allenjbaum stated that *"easy model build without dependencies that require sudo access"* is one of the two most important issues facing Sail at the moment.  Could you bullet-point the core requirements tfor this? Then it would be easier to conceive of a solution.

abukharmeh, 2023-02-10:
I think it could be really helpful if we make the CI upload the built model so people who are not intending to develop the the model can just download it and use it

martinberger, 2023-02-10:
> upload the built model so ...

Isn't that trivial?

abukharmeh, 2023-02-10:
Yep, just make the build directory job artifact

Timmmm, 2024-05-07:
I agree, I think uploading the generated C would be a good first move. Then all you need is a C compiler and libgmp which is not that hard to install.

This is how we work around the pain of installing the Sail compiler internally.

Alternatively (or ideally additionally), it would be nice if there was a binary distribution of Sail (including Z3!) that you could just download instead of having to deal with 🤬ing OPAM.

jordancarlin, 2025-02-11:
Seems like this can be closed now that we have a flow to distribute precompiled binaries (#715).

Timmmm, 2025-02-11:
Yes I think so, that and the Sail binary release. On Linux at least! Mac should come with the next Sail release and I'll do Windows when someone asks for it...

closed 2025-02-11 by Timmmm