riscv/sail-riscv #217: Extra Zero words in the end of signature file

algrobman, 2023-02-11:
Could you please comment about :

https://github.com/riscv-non-isa/riscv-arch-test/issues/312

?

jrtc27, 2023-02-11:
The Sail model just prints the words between begin_signature inclusive and end_signature exclusive. You can find the loop for yourself in c_emulator/riscv_sim.c, there’s really nothing fancy going on.

billmcspadden-riscv, 2023-02-11:
Support for misaligned is NOT a compile time option.  It is a runtime
option (-m, --enable-misaligned).
The Sail model will either trap on a misaligned address (if -m or
--enabled-aligned)  or it
won't (and the Sail model will do complete access to all parts).

Bill Mc.

abukharmeh, 2023-02-12:
I wonder if this is related to https://github.com/riscv/sail-riscv/pull/136#discussion_r772195502 ?

jrtc27, 2023-02-12:
No. The start is specified as being 16 byte aligned, and the signature is always a whole number of words.

algrobman, 2023-02-12:
First I see that to make some tests to pass I had to include "end" address for my DUT simulation model, which  always holds zeros.(?) (originally my signature dump loop was like:
for (addr = start_sig; addr < end_sig;  addr += 4) $display(mem[addr];
but I had to replace '<' by '<=' to match sail signature.

**Another question I have - did anybody run this simulator on RedHat 7 Linux or only on Ubuntu?**

jrtc27, 2023-02-12:
> First I see that to make some tests to pass I had to include "end" address for my DUT simulation model, which always holds zeros.(?) (originally my signature dump loop was like: for (addr = [...]

That doesn't make sense. `<` vs `<=` would only add *one* additional word, but in the issue you link to you're missing *two* words. Also, the current code in the Sail model is:

https://github.com/riscv/sail-riscv/blob/9547a30bf84572c458476591b569a95f5232c1c7/c_emulator/riscv_sim.c#L590-L597

which is clearly doing the same as your original `for (addr = start_sig; addr < end_sig; addr += 4)`.

> **Another question I have - did anybody run this simulator on RedHat 7 Linux or only on Ubuntu?**

Probably someone out there is/has given the ubiquity of old versions of RHEL in the hardware design world, but those of us who aren't in that world typically use OSes that are less than 8 years old. It's pretty unlikely that the OS you're running on is changing anything about the model's behaviour here, though, it's quite boring C code.

jrtc27, 2023-02-12:
Not a bug in the Sail model; figured out what was going on in https://github.com/riscv-non-isa/riscv-arch-test/issues/312#issuecomment-1426917948.

closed 2023-02-12 by jrtc27

algrobman, 2023-02-12:
Jessica,

Thanks a lot for the tip - I was looking values for different labels ... rvtest_sig_begin/rvtest_sig_end

algrobman, 2023-02-12:
yes,  this riscof is quite a mess!