riscv/sail-riscv #230: Request for Assistance: Running Programs on riscv_sim_RV64 with riscv-gnu-toolchain

FurtiveGreenhorn, 2023-03-14:
Hello, I would appreciate some advice on executing programs compiled with the riscv-gnu-toolchain on riscv_sim_RV64 and benchmarking them. I have tried two methods so far, but I am not very familiar with the htif mechanism. As a workaround, I used riscv-test to generate executable files, but unfortunately, it does not seem to support the clibrary and system calls. I am wondering if there are any supported projects available, or if I need to implement this functionality myself. I also attempted to run the programs on an OS, but it seems that the current sail os-boot does not support console input. I am not sure how to load the program and execute it properly. Do you have any suggestions or relevant implementations that I can refer to? Thank you for your help and guidance.

billmcspadden-riscv, 2023-03-14:
Hello FurtiveGreenhorn.  I'm going to steer you to my working branch which contains a couple of examples that you might find useful: [Cookbook](https://github.com/billmcspadden-riscv/sail-riscv/tree/master/cookbook/functional_code_examples)

The first directory to look at is br_j_asm/.   These are the files you want to look at:

test.S :  This is an assembly language program that was used as a rudimentary test of the 
e-trace specification and implementation.  It uses the htif mechanism.  Look for the symbols
j_target_end_pass and j_target_end_fail for examples of how to exit with a pass or fail flag.
The symbol, tohost, is the address to write.  It gets located in the .ld file

riscv_test.ld :  this is a general purpose ld file.  For your purposes,  it defines where "tohost" 
gets located.

Makefile:   the recipe for building an executable from assemblyh source.

I hope this helps.

Note: I'm currently in the process of improving the configuration capability of the RISC-V model using the RISCV-Config YAML files.  You may safely ignore those.

FurtiveGreenhorn, 2023-03-14:
Hi Bill McSpadden, thank you for providing me with these valuable resources. I'll take a look at the examples you suggested in the br_j_asm/ directory and see if they can help with my project. I appreciate your assistance!

ptomsich, 2023-03-14:
The first question should be to clarify what the intention of "on executing programs ... and benchmarking them." is.

SAIL is not a performance model: this would be a benchmark of the SAIL model/runtime and the host processor executing them, not of the RISC-V program running on any RISC-V hardware.

billmcspadden-riscv, 2023-03-14:
Philipp is correct.

My answer was to primarily address the question about the use of htif (tohost) to signal the
completion of a program when running a program against the Sail model. (The same mechanism
is used for Spike.)

ptomsich, 2023-03-14:
Regarding HTIF, here is glue code from our developer-test framework (and yes, this is compiled as a C++ file ... therefore the extern "C" linkages):
```c++
#include <stddef.h>
#include <stdint.h>
#include "putchar.h"

// HTIF 'IO ports' (SAIL/Spike will hook these symbols from the ELF)
volatile unsigned long tohost;
volatile unsigned long fromhost;

// HTIF devices
const uint8_t HTIF_SYSCALL = 0;
const uint8_t HTIF_TERMINAL = 1;

// HTIF device commands
const uint8_t HTIF_TERMINAL_READ = 1;
const uint8_t HTIF_TERMINAL_WRITE = 0;   // Not implemented in SAIL

static unsigned long htif_cmd(uint8_t device, uint8_t cmd, uint64_t payload)
{
  /*
    Spike's HTIF device interface.

    bitfield htif_cmd : bits(64) = {
    device  : 63 .. 56,
    cmd     : 55 .. 48,
    payload : 47 .. 0
    }
  */

  return ((uint64_t)device << 56) | ((uint64_t)cmd << 48) | (payload << 16 >> 16);
}

extern "C" void exit(int status)
{
  tohost = htif_cmd(HTIF_SYSCALL, 0, status << 1 | 1);
}

extern "C" int putchar(int c)
{
  tohost = htif_cmd(HTIF_TERMINAL, HTIF_TERMINAL_READ, (unsigned char)c);
  return (unsigned char)c;
}
```

FurtiveGreenhorn, 2023-03-14:
Thank you for explaining that. I'm currently considering the possibility of adding a performance model to SAIL, but I'm not entirely certain whether this would be helpful for the design of the instruction set or the processor. I was wondering if you could provide some insights or suggestions on this matter.

FurtiveGreenhorn, 2023-03-14:
Thank you for your explanation. I am currently attempting to add performance modeling to SAIL, but I am not entirely certain if this would be beneficial for the design of the instruction set or processor.

billmcspadden-riscv, 2023-03-14:
The Sail model has not been,  and never will be,  a suitable tool for performance measurements.
We have no semblance of timing and/or temporality (*). There is no clock.  The only things we know 
are 1) the sequential execution of instructions  2) the architecturally visible state of the core at 
instruction retirement (single instruction retirement only)  3)  at some point in time,  asynchronous 
traps are handled (non-deterministically). 

(*)  This is a little bit of a lie.  mtimer is actually implemented,  but the connection to instruction 
retirement is purely accidental.  Some day,  we may make mtimer a bit more useful,  but today
it is mostly useful in testing interrupt behaviour.

FurtiveGreenhorn, 2023-03-14:
Thank you very much for your explanation and your time. I appreciate your help in clarifying this for me.

closed 2023-03-14 by FurtiveGreenhorn

jrtc27, 2023-03-14:
> Regarding HTIF, here is glue code from our developer-test framework (and yes, this is compiled as a C++ file ... therefore the extern "C" linkages):
>
> ```c++
> #include <stddef.h>
> #include <stdint.h> [...]

Shouldn’t a bunch of your unsigned longs be uint64_ts? IIRC everything is 64-bit still on RV32.

ptomsich, 2023-03-14:
I just checked the SAIL source and you are right.
Just committed a fix to our internal repository.

abukharmeh, 2023-03-15:
On 32-bit, HTIF does not have BCD as HTIF packets should be zero extended in theory, but that's not the current behavior in the model !  https://github.com/riscv-software-src/riscv-isa-sim/issues/364#issuecomment-607657754