riscv/sail-riscv #231: Error when making C/OCaml emulators after pulling new commits

XinlaiWan, 2023-03-19:
There is an error when making emulators in the master branch now. I'm not sure if there are problems in my Sail environment or other issues. Could someone help identify the issue?

```
Syntax error:
[model/prelude.sail]:111:10-10
111 |val not = pure {coq: "negb", _: "not"} : forall ('p : Bool). bool('p) -> bool(not('p))
    |          ^
    | current token: pure
make: *** [Makefile:202: generated_definitions/ocaml/RV64/riscv.ml] Error 1
```
```
Syntax error:
[model/prelude.sail]:111:10-10
111 |val not = pure {coq: "negb", _: "not"} : forall ('p : Bool). bool('p) -> bool(not('p))
    |          ^
    | current token: pure
make: *** [Makefile:236: generated_definitions/c/riscv_model_RV64.c] Error 1
```

jrtc27, 2023-03-19:
The GitHub Actions build from 5 days ago built fine. What version of Sail are you using?

jrtc27, 2023-03-19:
Probably you have an older version; 0.15 is required to support the pure keyword there.

XinlaiWan, 2023-03-19:
Thank you! I was using Sail 0.14. No error occurs when I upgrade to Sail 0.15.

closed 2023-03-19 by XinlaiWan