riscv/sail-riscv #249: Where does come c_emulator /SoftFloat-3e/

Et7f3, 2023-04-08:
This [folder](https://github.com/riscv/sail-riscv/tree/0.5/c_emulator/SoftFloat-3e) seems to contains vendored copy of somewhere. I wanted to do a PR that affect this file but it will be lost on next update. So can you say where it does come from ? A quick search showed https://github.com/ucb-bar/berkeley-softfloat-3 and you added RISVC folder (Can you consider upstream this change ?). Am I right ?

ptomsich, 2023-04-08:
Yes, this is based on the Berkeley SoftFloat library.

However: What change do you plan to make and why? There is no issue open against FP in the SAIL model…
Also note that the long-term plan is to replace the SoftFloat library with pure SAIL implementations of the FP functionality (as this will be a required to use SAIL as part of the specifications).

Et7f3, 2023-04-08:
> What change do you plan to make and why?

Minor fix so I can play with it on MacOS https://github.com/riscv/sail-riscv/pull/250 If `make check` works on Darwin it should have the same behavior than linux ?

aamartin0000, 2023-05-15:
Philipp, perhaps this should be a different topic, but regarding the long-term plan to replace SoftFloat with a SAIL implementation, will you guarantee that the behavior will be exactly the same?  In other words, if these two produce different results, which is considered golden?

There are a number of different FP hardware implementations (including in github), and in addition to ISA differences (e.g. RISC-V vs. x86 vs. ARM), there are discrepancies in results and exception flags in these various implementations, mostly due to the way rounding and subnormals are handled.  I have been working on a hardware implementation based on HardFloat partially because it's the most compatible with SoftFloat (which is what we're using in our simulators).  I don't want to have to modify both the hardware implementation and software simulator if the SAIL definition is different.

ptomsich, 2023-05-15:
Given that RISC-V is heading towards using SAIL as the "executable specification", we will eventually need an FP implementation that is pure SAIL (and does not rely on any magical devices).  So the expectation is that (once all bugs have been ironed out) the SAIL implementation will be the authoritative one.

If the current soft-float implementation were to deviate from the intent of the RISC-V unprivileged specification, the specification would win (and the current soft-float implementation would get fixed).  So changing the implementation is just a sideway move, although one that will allow the FP semantics to be fully included as SAIL code in the specification documents.

aamartin0000, 2023-05-15:
Understood.  What I'm asking about is, particularly while this transition is underway, there will inevitably be some cases that SoftFloat and SAIL disagree.  I know this is probably on a case-by-case basis, but **who will arbitrate which one is producing the correct result**?  I'm obviously leaning toward SoftFloat being correct, since it "got there first" and is in use in implementations in the wild, but I also agree that the "correct" result should win the day.

Please keep the FP SIG in the loop when this transition is underway.

ptomsich, 2023-05-16:
/cc @billmcspadden-riscv

jrtc27, 2023-05-16:
I assume like any software project of this kind the new version would be on a feature branch or on the main branch behind an option, with it only being made the default on the main branch once equivalence has been reached and tested. Ideally on a branch to keep noise down in the commit history.

KotorinMinami, 2024-06-28:
With #250 already merged, are there any remaining issues to be resolved , or can this issue be closed ?

closed 2024-06-28 by Timmmm