riscv/sail-riscv #254: Cannot compile due to syntax error in prelude.sail

ved-rivos, 2023-04-16:
I have installed sail using process documented in:
https://riscof.readthedocs.io/en/stable/installation.html

```
$ sail -v
Sail 0.14 (sail2 @ opam)
```

I get the following error on hitting make:
```

Syntax error:
[model/prelude.sail]:111:10-10
111 |val not = pure {coq: "negb", _: "not"} : forall ('p : Bool). bool('p) -> bool(not('p))
    |          ^
    | current token: pure
make: *** [Makefile:202: generated_definitions/ocaml/RV64/riscv.ml] Error 1
```
Please help. If there is a better manual for installing and running sail then please do point to that.

ved-rivos, 2023-04-16:
This was the only instance of `pure` keyword I could find in this repo and looking through the sail manual (01/18/23 version) I was not able to understand how this keyword should be used. Please also help me understand what the `pure` keyword does.

jrtc27, 2023-04-16:
Your version of sail is too old; that documentation (which is maintained by RISCOF people, not us) is presumably outdated. You need Sail 0.15.

jrtc27, 2023-04-16:
https://github.com/rems-project/sail/blob/sail2/INSTALL.md should have up-to-date installation instructions for Sail; it claims to need OCaml 4.08 or above, which explains why installing Sail from a 4.06.1 environment doesn't get you Sail 0.15.

closed 2023-04-16 by jrtc27