riscv/sail-riscv #268: fcvtmod.w.d produces incorrect floating point exception flags for inputs with exp == 31

mlawson-tt, 2023-05-31:
Executing the instruction `fcvtmod.w.d x31, f31` with a value of -4108055294.3100147 (hex value 0xc1ee9b7e5fc9eba4) in f31 produces an inexact FP exception in SAIL but produces an invalid FP exception in Spike.  Both seem to produce the same integer value that is stored in x31 (186912002 / 0xb240d02).  Given that the input FP value is outside the range of 32-bit signed integers and that the input value is entirely different than the output value (i.e. not just due to the fraction truncation), it appears an overflow occurs in this situation, and thus Spike is correct in that an invalid FP exception should be raised instead of inexact.

Note that the post-bias exponent in this case is 31, so it does not trigger the exponent-larger-than-31 condition needed to raise the invalid exception.  Spike has an additional explicit check for when the pre-truncated value exceeds the limits of an int32 and raises the invalid exception when this check is met, so it seems something similar may be necessary here.

ptomsich, 2023-05-31:
Taking a second look at the input value, it looks like the logic in SAIL is indeed flawed, as it does not consider that we have a signed input and a signed result: we thus shift the implicit bit into the position of the sign-bit.

I'll look at the required logic changes for the SAIL.

charmitro, 2023-05-31:
> We may be off-by-one on the check post-bias exponent (i.e., the explicit check is supposed to ascertain that the implicit bit does not get shifted out), but may not have a test-case on the ACTs to check for this.
> 
> @charmitro Does the test-coverage catch all the corner-cases of fcvtmod.w.d? If not, (besides fixing it here) we will need to address this in the ACT PR that is pending…

I'll address this in the ACT PR as you said. It seems that a couple of corner-cases can be added.

ptomsich, 2023-05-31:
@mlawson-tt The produced integer bits may overflow a signed 32bit (prior to applying the sign-bit, we can have 0..0x8000_0000 for negative numbers and 0..0x7ff_ffff for positive).  Anything else should raise NV.

The cleanest way to express this might be a signed fixed point representation (in two's complement) before multiplying in the exponent... I need to think about this before proposing a fix.

PR #442 (2024-04-29): Correct fcvtmod.w.d flag generation logic for exp == 31
Fixes #388 and fixes #268. fcvtmod.w.d is producing an inexact instead of invalid flag for numbers with an exponent of 31 that are not representable as a two's complement integer (> 2^31 - 1 or < -2^31).

Per #388, 

> According to the Zfa specification, "Floating-point exception flags are raised the same as they would be for FCVT.W.D with the same input operand." According to the unprivileged specification Table 11.4, the invalid exception should be set for inputs > 2^31 - 1 or < -2^31.

This is resolved by adding an additional check whe [... cut]
closed 2024-04-29 by billmcspadden-riscv