riscv/sail-riscv #270: Vector extension VLEN configuration

XinlaiWan, 2023-06-05:
In RISC-V Vector extension, there are "design-time constants" like VLEN and ELEN. They will not be modified during execution of the model. The current implementation defines them as registers in Sail code so that they can be configured at the beginning of simulation by command line options or other configuration approaches. Currently we don't have such a configuration interface to let the user specify these values and assign them to the registers, so we have to manually set them in the code before compiling the model with RVV support. I'm opening this issue to explain this situation, and there is also a comment in [`riscv_vlen.sail`](https://github.com/riscv/sail-riscv/blob/vector-dev/model/riscv_vlen.sail) of `vector-dev` branch as a note.

jordancarlin, 2024-10-19:
Mostly resolved with #590. Additional configuration will be added with the upcoming full model configuration system.

Timmmm, 2024-10-19:
Yeah I think we can close this.

closed 2024-10-19 by Timmmm