riscv/sail-riscv #284: PMP NAPOT model is misbehaving

UmerShahidengr, 2023-07-14:
While testing a basic code in S-mode, I and @allenjbaum have noticed peculiar behavior in PMP NAPOT mode. I am dragging through the problem with the following example:

We tried going from M-mode to S-mode using RVTEST_GOTO_LOWER_MODE macro as defined in riscv-arch-test trap handler. Upon reaching to the S-mode, the program started giving fetch-access-fault on every instruction, so we thought it was the due to PMP configurations. So we did the following configurations:
1. Set up a PMP region in TOR mode and provide full physical space to it. That resolved our issue of fetch-access-fault and the code worked in S-mode. For reference, we used the following code:
```
li x1, -1
csrw pmpaddr0, x1  // Added 0xFFF...FF to pmpaddr0
li x1, 0x0F
csrw pmpcfg0, x1  // configure pmpcfg0 into TOR with RWX enabled

```
2. We removed the TOR mode code and tried NAPOT mode to do the same thing. We setup a PMP region in NAPOT and assign full physical space to that mode as well. It should have behaved the same way as that of TOR mode (and Spike behaved the same way) but Sail started giving fetch-access-fault and did not define the PMP region correctly. For reference, here is the code we tried before going into S-mode:
```
li x1, -1
srli x1, x1, 2
csrw pmpaddr0, x1 // Added 0xFFF...FF << 2 into pmpaddr0
li x1, 0x1F
csrw pmpcfg0, x1  // configured pmpcfg0 into NAPOT mode with PWX enabled

```
Sail is showing two different behaviors for two different PMP regions (TOR and NAPOT) under the similar configurations, I think NAPOT mode has some bug which is needed to be fixed as Spike gave the expected result in both cases.

KotorinMinami, 2024-08-12:
I have taken a look about this issue.

When using both modes to configure the physical space and go into S-mode, it both work well. However, I'm not certain whether the test file can cover all the scenarios mentioned in this issue.

Here is my [test file](https://github.com/KotorinMinami/plct-working/tree/main/sail-riscv/tests_NAPOT) - could you assess whether the bug has been resolved?

Timmmm, 2024-08-12:
This could be because there was a bug in how addresses above 2^32 were handled on RV32 and it [was fixed recently](https://github.com/riscv/sail-riscv/commit/a71fee30035e6d504b156cda0802902345766e70). Maybe worth trying just before/after that commit if you still have everything set up and fresh in your mind @KotorinMinami.

Otherwise I'd guess we need a complete reproducer for this to do any more.

KotorinMinami, 2024-08-19:
@Timmmm I would be more than happy to do that. When I checked out the code before the commit and compiled it with,

```bash
ARCH=RV32 make csim
```
 also compiling the test_NAPOT file to the RV32 version, I encountered a fetch-access-fault when I ran the code. However, after checking out the code post-commit and repeating the process, it ran successfully. 
```bash
riscv_sim_RV32 --pmp-count=64 elf/test_NAPOT.elf
```
So, I believe this issue may have been resolved. Here are my [code and binary files](https://github.com/KotorinMinami/plct-working/tree/main/sail-riscv/tests_NAPOT/RV32); what do you think?

Timmmm, 2024-08-19:
Sounds conclusive to me - great work! I'll close this. @UmerShahidengr @allenjbaum if you manage to reproduce the issue on `master` feel free to reopen.

closed 2024-08-19 by Timmmm