riscv/sail-riscv #288: Adding a New Function in C and Calling it from a Sail File

ryan037, 2023-08-04:
I want to add a new function in a C file and then call it within a Sail file. Just like in the **c_emulator/riscv_platform.c** file, a function is defined as follows:

```
mach_bits plat_clint_base(unit u) {
  return rv_clint_base;
}
```
In the **model/riscv_platform.sail** file, it is declared with:

`val plat_clint_base = {ocaml: "Platform.clint_base", interpreter: "Platform.clint_base", c: "plat_clint_base", lem: "plat_clint_base"} : unit -> xlenbits`

and **plat_clint_base** is called later in **riscv_platform.sail**. I want to do the same thing as described above. How should I do about it?

billmcspadden-riscv, 2023-08-04:
Hi Ryan.  Thanks for the question.  Let me try to answer.

There's two answers to the question.  First,  how does the Sail language
handle
the calling of a C function.  Second,  how do we extend the RISC-V Sail
model
to make use of this functionality.

For the first question,  take a look at....
https://github.com/billmcspadden-riscv/sail/blob/cookbook_br/cookbook/doc/TheSailCookbook_Complete.pdf
Martin Burger and I put together a cookbook with a bunch of
tutorial examples
that answer these sorts of questions.  Take a look at chapter 12.  Both
Martin
and I put in an example of how to do this.

For the second question,  you are already on the right track.  Take what
you learned
from answering the first question,  and then see how things are done for
the RISC-V
Sail model.  As you have seen,  there are many examples.  Note: You are
not required to
supply an ocaml | interpreter | lem method in order to compile the C
model.

Let me know if you get stuck.  Both Martin and I have exercised this
interface quite
a bit.

Bill Mc.

ryan037, 2023-08-05:
Thank you for answering my questions.

closed 2023-08-05 by ryan037