riscv/sail-riscv #292: sail compile temp broken.

dansmathers, 2023-08-16:
When I ran:
make ARCH=RV32

echo "declare {isabelle} rename field sync_exception_ext = sync_exception_ext_exception" >> generated_definitions/lem/RV32/riscv_types.lem
lem -isa -outdir generated_definitions/isabelle/RV32 -lib Sail=/home/504984/.opam/default/share/sail/src/lem_interp -lib Sail=/home/504984/.opam/default/share/sail/src/gen_lib \
      handwritten_support/0.11/riscv_extras.lem handwritten_support/0.11/mem_metadata.lem handwritten_support/0.11/riscv_extras_fdext.lem \
      generated_definitions/lem/RV32/riscv_types.lem \
      generated_definitions/lem/RV32/riscv.lem
File "generated_definitions/lem/RV32/riscv.lem", line 282, character 24 to line 282, character 34
  Type error: unbound variable for targets {isabelle}: sign_extend
make: *** [Makefile:318: generated_definitions/isabelle/RV32/Riscv.thy] Error 1

Bill is investigating.  The current workaround is to only build the C simulator:
make [options] csim

I verified 
make ARCH=RV32 csim
worked for me.
creating an issue for tracking

bacam, 2023-08-17:
Sounds like the name shadowing bug I fixed in Sail the other day: https://github.com/rems-project/sail/commit/1b7964d85ac179623d6d2e497b38407ad77fd425

mkpest, 2023-08-24:
It seems like [last commit - 58cac61](https://github.com/riscv/sail-riscv/commit/58cac61d9ddde591902c933a9dfa5d8ba3fca6da) - broke `make`. Running `make` from [ae905fb](https://github.com/riscv/sail-riscv/commit/ae905fb888cbb21c782bacf86be182d9e20b8895) works.

billmcspadden-riscv, 2023-08-24:
I was able to recreate this after some difficulty.  

I see the problem with Sail version 0.15,  not 0.14.

Questions:

1. To fix this problem,  we will need to do update the version of Sail, yes?
Or we might back out PR #279 (not preferred).  

2. Why wasn't this caught during CI?  I'm assuming that the CI flow did
not use Sail version 0.15, else this would have been caught.

3.  @bacam, did you check to see if your change to Sail fixed this problem?
(I'm attempting to check,  but am having serious problems getting opam to do what I want.)

bacam, 2023-08-25:
It does (I was looking at sail-cheri-riscv at the time, but it works with the main sail-riscv repo too).  A workaround would be to rename `sign_extend` and `zero_extend` to avoid the name clash with Lem's library, although it might be simpler to remove the theorem prover models from the default target, and only build them with the latest version of Sail in the CI.

jrtc27, 2023-08-25:
CI only builds the emulators; see test/run_tests.sh

Timmmm, 2024-05-07:
Sounds like this is fixed?