riscv/sail-riscv #295: mideleg MEI MTI MSI bits are read only zero but maybe shouldn't be?

Timmmm, 2023-08-22:
The model currently has this code:

```
function legalize_mideleg(o : Minterrupts, v : xlenbits) -> Minterrupts = {
  /* M-mode interrupt delegation bits "should" be hardwired to 0. */
  /* FIXME: needs verification against eventual spec language. */
  let m = Mk_Minterrupts(v);
  let m = update_MEI(m, 0b0);
  let m = update_MTI(m, 0b0);
  let m = update_MSI(m, 0b0);
  m
}
```

But there doesn't appear to be anything in the spec that requires this. I can't find the "should" that the comment is referencing. It could possibly be a misinterpretation of this text:

> Similarly, an implementation shall not fix as read-only one any bits of mideleg corresponding to machine-level interrupts (but may do so for lower-level interrupts).

Indeximal, 2023-09-30:
I'm currently struggling to understand this aswell. My explaination would be that it should (implicitly) be impossible to handle a machine mode interrupt in a non-machine mode.
Given the delegation mechanism exists, I not quite sure why then there would even be separate interrupts defined per level.

jrtc27, 2023-09-30:
https://github.com/riscv/sail-riscv/commit/cb0fdf1bbe09e2ce99a0ba6a3f9967f8624423f3 is where that came from, for what it's worth. The model was a *lot* less complete and correct back then. Perhaps it was a reflection of the configuration the model was implementing at the time, or perhaps a misunderstanding in whatever communication that was. You'd have to ask Prashanth and/or Andrew, because I can't obviously find such a conversation in GitHub issues or the isa-dev list.

Indeximal, 2023-09-30:
It seems implementation dependent if I look the comment in section 4.1.3 of the priviledged isa manual ("Supervisor Interrupt Registers (sip and sie)").
> Bits 3, 7, and 11 of sip and sie correspond to the machine-mode software, timer, and
external interrupts, respectively. Since most platforms will choose not to make these interrupts
delegatable from M-mode to S-mode, they are shown as 0 in Figures 4.6 and 4.7.

allenjbaum, 2023-10-01:
Technically, all interrupts and exceptions are machine mode traps unless
delegated.
 - though they can't be delegated if the trap is from Mmode, and
 - they can't be delegated if the corresponding delegation bit must be
RdOnly 0 (I believe they cannot be RdOnly 1)
The very last line of 3.1.8 says "in particular, medeleg[11] is read-only
zero."

There is no such comment regarding  interrupt delegation, so all interrupts
may be delegated.

But, the overriding confusion here is the name. There are *no* mode
specific interrupts.
Machine External Interrupt and Supervisor External Interrupt are identical
except for their name, and their priorities.
Either, both, or neither can be delegated.
Think of them simply as External Interrupt1 and External Interrupt2; The
names only reflect how it was expected they would be used, not how they
must be used.

Timmmm, 2023-10-08:
So are we in agreement that the code should be this?

    function legalize_mideleg(o : Minterrupts, v : xlenbits) -> Minterrupts = Mk_Minterrupts(v)

Timmmm, 2023-10-19:
Ok I've read the spec more and I think the code is actually fine. It's just implementing the most likely option out of all possible implementations.

One day `riscv-config` will let you say exactly which bits are delegable to S mode.

closed 2023-10-19 by Timmmm