riscv/sail-riscv #296: pmpcfg allows illegal values R=0, W=1

Timmmm, 2023-08-22:
For the `pmpcfg` registers the spec says:

> The R, W, and X fields form a collective WARL field for which the combinations with R=0 and W=1 are reserved.

However the current Sail code allows you to write any value and doesn't legalize RWX to prevent R=0, W=1.

The relevant functions are `pmpWriteCfgReg()` and `pmpWriteCfg()`.

Pagerd, 2024-06-18:
Seems that the issue is in the function pmpCheckRWX?  looks like add an R judgment can fix it. Do I need to add a PR to fix it?

Timmmm, 2024-06-18:
We've got the code for this already actually - just didn't make it to the top of my upstreaming list yet. I'll try and do it this week.

Timmmm, 2024-06-18:
Actually no need - the code for this was [already in the PMP update](https://github.com/riscv/sail-riscv/commit/4de2bff12d967d91dd064e4a49e25ca4785f25e3#diff-3d251956b0e0495882279272a39852422ce72366f9ca0d4b6f337c8489691444R169).

closed 2024-06-18 by Timmmm