riscv/sail-riscv #307: SV48 TLB Issue

YazanHussnain-10x, 2023-09-22:
## Issue Description

I write the architectural verification test for **SV48**, which you can find [here](https://github.com/YazanHussnain-10x/su21-lab-starter/tree/main/SV48_tests). I used the sail as my reference model. When I checked the sail's log for the test, I noticed that when address translation is enabled, the page table is correctly accessed for the first instruction. However, for the next instruction, which is on the same page, the page table should not be accessed because the TLB should be updated with the translation info as per the specs. The translation is expected to be found in the TLB. Unfortunately, in the case of **SV48**, the TLB is not accessed. Instead, for every instruction, the sail is accessing the page table, which should not be happening.
I also write a test for **SV39**, TLB in that test is working as expected.

Here is the part of the log when address translation is enabled:
``` C
[486] [M]: 0x008000000000087C (0x341494F3) csrrw s1, mepc, s1
CSR mepc -> 0x0000000000000000
CSR mepc <- 0x0000020000000884 (input: 0x0000020000000884)
x9 <- 0x0000000000000000

mem[X,0x0080000000000880] -> 0x0073
mem[X,0x0080000000000882] -> 0x3020
[487] [M]: 0x0080000000000880 (0x30200073) mret
CSR mstatus <- 0x0000000A00000080
ret-ing from M to S

mem[R,0x0080000000006020] -> 0x00200000000000CF // Page Table Accessed
mem[X,0x0080000000000884] -> 0x0293
mem[R,0x0080000000006020] -> 0x00200000000000CF // Page Table Accessed Again
mem[X,0x0080000000000886] -> 0x0050
[488] [S]: 0x0000020000000884 (0x00500293) addi t0, zero, 5
x5 <- 0x0000000000000005

mem[R,0x0080000000006020] -> 0x00200000000000CF // Page Table Accessed Again
mem[X,0x0080000000000888] -> 0x9293
mem[R,0x0080000000006020] -> 0x00200000000000CF // Page Table Accessed Again
mem[X,0x008000000000088A] -> 0x0272
[489] [S]: 0x0000020000000888 (0x02729293) slli t0, t0, 39
x5 <- 0x0000028000000000
```

As you can see, the page table is accessed, and the PTEs at all levels (in this case only one page table is used) are read again for the second instruction after mret.

## Required Changes

Add support for TLB lookup and updating for the SV48 address translation scheme.

## Current Status

Currently TLB for **SV48** is not working.

allenjbaum, 2023-09-22:
This may be a performance problem, but it should be OK from a correctness
perspective.

rsnikhil, 2023-10-31:
As @allenjbaum said above, TLBs are only intended for performance improvement, and do not impact functional correctness.

Currently only the Sail code for Sv32 and Sv39 include a TLB (and just a single-entry TLB), whereas The Sail code for Sv48 does not have a TLB at all.  So, in traces with Sv48 you will not see any TLB accesses.

Most tests that are run in sail-riscv are small, where this performance of vm-to-pa is not noticeable.  I believe TLBs were added only for Sv32 and Sv39 because those were the only two Linux-boots tested, and performance is notable for Linux boot.

@rsnikhil is currently reworking the Virtual-Memory translation code to unify Sv32, Sv39 and Sv48 translations into common functions, and hopefully this will cover TLBs as well.

Timmmm, 2024-05-07:
This should be "fixed" now that the new virtual memory code is merged. Feel free to reopen if not!

closed 2024-05-07 by Timmmm