riscv/sail-riscv #311: Use of ext_* read/wrtite hooks for standard extensions

defermelowie, 2023-09-26:
Question about [CODE_STYLE.md](https://github.com/riscv/sail-riscv/blob/master/CODE_STYLE.md): At the bottom it states that `ext*` types and hooks should not be used for standard extensions. However, [ExtendingGuide.md](https://github.com/riscv/sail-riscv/blob/master/doc/ExtendingGuide.md) claims that "access to newly added CSRs can be hooked in `riscv_csr_ext.sail`". Furthermore, the F and D standard extensions make heavy use of the `ext_*` hooks in [`riscv_fdext_control.sail`](https://github.com/riscv/sail-riscv/blob/master/model/riscv_fdext_control.sail).

Is the statement from CODE_STYLE.md still valid? If this is the case, how should CSR accessors of standard extensions then be defined? Directly in [`riscv_insts_zicsr.sail`](https://github.com/riscv/sail-riscv/blob/master/model/riscv_insts_zicsr.sail) or in some other way?

jordancarlin, 2024-10-16:
Now that #543 is merged, the documentation can be updated and then this can be closed.

jrtc27, 2024-10-16:
The CSR case was an exception in the past because it was a scattered function and thus one could add new CSRs without modifying existing files. #543 renamed things to drop the ext_ prefix to reflect that it wasn't really appropriate naming here. Similarly ext_write_fcsr is inappropriately named. The intent is that all the ext_ functions are stub hooks that out-of-tree extensions can overwrite to do something useful, and are included in the model based on what such extensions need, with CHERI being the historical motivating use case.

Timmmm, 2025-09-16:
I think this is solved is solved now, though the `ExtendingGuide.md` is now out of date since we renamed all of the files.

closed 2025-09-16 by Timmmm