riscv/sail-riscv #319: Support multiple spec versions

Timmmm, 2023-10-10:
Currently the model doesn't really say what version of any of the specifications it is implementing. We need to add functions in the model to allow switching on the versions.

In the medium term these can be controlled by command line flags. In the long term they can come from the riscv-config YAML. That [has entries for the privileged and unprivileged specs](https://riscv-config.readthedocs.io/en/stable/yaml-specs.html#user-spec-version) which they have confusingly called `User_Spec_Version` for some reason.

More spec versions are required for all the extensions that are supported (e.g. bitmanip, vector). Those two would be a start though.

allenjbaum, 2023-10-10:
Officially, there are no spec versions, just extensions - except the priv
spec has extension names with numbers in  them. (e.g. priv 1.11, 1.12, and
eventually 1.13 I suppose).
There are some chips out there that have implemented unratified versions of
specs - those are considered custom extensions.
Whether we want to support those in Sail is a different discussion.

billmcspadden-riscv, 2024-08-01:
I've assigned myself this issue.

I'll be working on PRs for implementation of priv 1.13 features.

Timmmm, 2024-08-01:
I think it's probably a good idea to add the mechanism for specifying the priv/unpriv spec version (i.e. a version enum, checking function and command line flags) before adding any new features. Also 1.13 doesn't seem to exist - it uses date based versions now apparently.