riscv/sail-riscv #323: Enormous regression in compile time (x10) and memory use (x4) due to Vector extension

Timmmm, 2023-10-23:
Unfortunately the vector extension has caused the model compile time and memory usage to balloon. Here's a benchmark on my laptop (a mid range Dell one running Linux):

| Vector? | No | Yes |
|-------------|--|--|
|Commit | c04cf29c2215ff614a83ac483b9545a995adca65 | c90cf2e6eff5fa4ef7b93cc2020166dea7453fc6 |
| Wall Time | 59s | 9m 34s |
| Max RSS | 2.2 GB | 7.9 GB |
| Generated C size | 8 MB | 19 MB (IIRC; I forgot to write this down)|

We compile 4 models in parallel, which was working great until now (10GB is not too much of an ask). 32 GB is though! So with this change we'd now have to probably compile one model at a time to avoid OOM, which increases the compile time from 1 minute to 40 minutes! A bit of a deal-breaker.

Timmmm, 2023-10-23:
I forgot to say, this is for

    ARCH=RV64 make c_emulator/riscv_sim_RV64

billmcspadden-riscv, 2023-10-23:
Wow.  This is not what I saw when I compiled the csim model.  I was seeing
compile times of maybe 4-5 minutes.

Are you compiling just the csim model?  Or are you also compiling the OCAML
model?

Bill Mc.

arichardson, 2023-10-23:
I am seeing 4 minutes for sail -> c and 1m10 for C->exe

Alasdair, 2023-10-23:
The source of the slowness is the string parsing parts of the assembly mappings

arichardson, 2023-10-23:
Here is what I see from perf:
```
+   64.87%     0.91%  sail     sail                  [.] Stdlib.List.map_236                                                                                                                                                           
+   58.32%     0.00%  sail     sail                  [.] caml_major_collection_slice                                                                                                                                                   
+   48.79%     8.26%  sail     sail                  [.] mark_slice                                                                                                                                                                    
+   46.70%     0.07%  sail     sail                  [.] Libsail.Util.string_of_list_1200                                                                                                                                              
+   46.32%     0.47%  sail     sail                  [.] Stdlib.^_140                                                                                                                                                                  
+   43.93%     0.30%  sail     sail                  [.] caml_alloc_string                                                                                                                                                             
+   43.23%     0.01%  sail     sail                  [.] caml_check_urgent_gc                                                                                                                                                          
+   40.50%    27.34%  sail     sail                  [.] mark_slice_darken.constprop.0                                                                                                                                                 
+   32.87%     0.19%  sail     sail                  [.] Libsail.Jib_util.full_string_of_ctyp_2814                                                                                                                                     
+   31.69%     0.00%  sail     sail                  [.] Libsail.Jib_util.ctyp_compare_3383                                                                                                                                            
+   29.14%     0.06%  sail     sail                  [.] Stdlib.List.fold_left_275                                                                                                                                                     
+   26.09%     0.14%  sail     sail                  [.] Stdlib.Set.add_189                                                                                                                                                            
+   22.45%     0.00%  sail     sail                  [.] Libsail.Jib_util.instrs_ctyps_4593                                                                                                                                            
+   21.52%     0.00%  sail     sail                  [.] caml_alloc_small_dispatch                                                                                                                                                     
+   21.32%     0.00%  sail     sail                  [.] caml_call_gc                                                                                                                                                                  
+   14.54%     0.01%  sail     sail                  [.] Stdlib.Set.union_292                                                                                                                                                          
+   13.20%    13.20%  sail     sail                  [.] caml_page_table_lookup                                                                                                                                                        
+    9.59%     0.00%  z3       libc.so.6             [.] __libc_start_call_main                                                                                                                                                        
+    9.51%     6.74%  sail     sail                  [.] sweep_slice                                                                                                                                                                   
+    9.48%     0.00%  sail     sail                  [.] Libsail.Jib_util.fun_6197                                                                                                                                                     
+    8.25%     0.03%  sail     sail                  [.] Libsail.Type_check.crule_4143                                                                                                                                                 
+    7.03%     0.24%  sail     sail                  [.] Libsail.Jib_util.map_instr_4296                                                                                                                                               
+    6.72%     2.03%  sail     sail                  [.] Libsail.Ast_util.compare_1484                                                                                                                                                 
[... cut ...]
+    6.41%     0.00%  sail     sail                  [.] caml_empty_minor_heap                                                                                                                                                         
+    6.28%     0.07%  sail     sail                  [.] Libsail.Type_check.check_exp_4355                                                                                                                                             
+    6.22%     0.00%  sail     sail                  [.] Sail.run_sail_2374                            
```

cfbolz, 2023-10-24:
of the 19 MB generated C code, about 2.4M are lines that look like this:

```c
append_64(&zghz33485, zghz33485, UINT64_C(0x0000000000000000));
```

Every line like this is repeated 1024 times. this seems to be done to initialize the vector register bitvector values (type `bits(65536)`). I don't understand why Sail does it with repeated bitvector appends, that seems very inefficient?

Alasdair, 2023-10-24:
We haven't had any `bits(65536)` before

Alasdair, 2023-10-24:
The large sequences of append_64 are generated by the `-Oconstant_fold` option. It'll see `zeros(<large number>)` and happily expand it to a literal. I'd need to add some kind of escape hatch to the constant folding mechanism so certain functions can cause it to bail out if they detect inputs that would lead to very large literals I think.

Alasdair, 2023-10-24:
https://github.com/rems-project/sail/pull/353 should be about a 2x improvement. Will probably try some memory profiling, as perf isn't being super helpful.

Timmmm, 2023-10-25:
@Alasdair that doesn't help much with compile time but it does massively reduce memory usage. Updated table:

| Version | No Vector | Vector | ...and latest `sail2` compiler) |...and `vlenmax`=1024 | 
|-------------|--|--|----|-|
| Wall Time | 59s | 9m 34s | 9m 45s | 11m 13s (?)|
| Max RSS | 2.2 GB | 7.9 GB | 4.4 GB | 2.3 GB |
| Generated C size | 8 MB | 19 MB?| 14 MB | 12MB |

So at least the memory usage is solved by reducing `vlenmax` and using Alasdair's patch. Just need to solve the compile time next...

Unfortunately my knowledge of profiling OCaml is about as much as my knowledge of Sail compiler internals...

Alasdair, 2023-10-25:
Are you running in a new virtual machine each time?

Timmmm, 2023-10-25:
No, just `make clean` then `ARCH=RV64 time -v -- make c_emulator/riscv_sim_RV64`. Also it's on a laptop, so I wouldn't read anything into the extra 2 minutes on the last run (no way should a shorter `vlenmax` make it slower...).

Alasdair, 2023-10-25:
Have you tried removing the `-Oconstant_fold` flag in the Makefile? With vector, on my MacBook that causes the linker to just take forever, although it seemed ok on Linux.

After removing the `-Oconstant_fold` flag, I have another Sail patch that reduces the time and memory usage on my laptop to 35s/900mb for the model with vector, and that's still just fixing various easy things.

Timmmm, 2023-10-25:
Yeah I did but it made things a bit worse IIRC, but maybe I did it wrong & my memory is hazy.

> that reduces the time and memory usage on my laptop to 35s/900mb for the model with vector, and that's still just fixing various easy things.

Amazing! 35s is pretty tolerable but if there's still low hanging fruit, faster is always better!

jrtc27, 2023-10-25:
How much performance is lost without -Oconstant_fold?

cfbolz, 2023-10-25:
another option might be to leave constant-folding on but have a special-case for bitvector constants filled with zeros in the backends

Timmmm, 2023-10-26:
I tried again with the latest `sail2`.

Running Sail takes 1m 36s and uses 1.8 GB of RAM.

Compiling the C code takes 8m 10s and uses 1.2 GB of RAM.

So it seems like the memory use has been totally solved, but compile time is still pretty bad! My guess is `zencdec_backwards()` which is now a 30k line function. Perhaps some outlining would help there? Maybe there could be a generic outlining pass that splits up huge functions?

Alasdair, 2023-10-26:
I haven't looked at the C too closely yet. I think the vector constant folding is the main culprit. The other easy improvement would probably be to split the generated C into multiple files to allow the C compiler to more effectively work in parallel.

For Sail I now have it down to 20s and 900MB-1GB of memory on a M1 MacBook Pro (without constant folding). I think 10s might be doable before trying to parallelise things there, which would require users to upgrade to OCaml 5. Constant folding is probably very expensive in Sail as well, and that might be harder to fix. The only incremental thing we do is remember SMT problems between runs, but obviously adding more caching would make repeated runs much faster (although cache-invalidation for any kind of incremental compilation will always be tricky).

Timmmm, 2023-10-27:
> The other easy improvement would probably be to split the generated C into multiple files to allow the C compiler to more effectively work in parallel.

I'd probably try and avoid this if it's solvable in a different way because:

1. It might cause slower runtime because of less optimisation. Unless you use `-flto` (which `sail-riscv` does) but then you don't get any compile time benefit. Not an issue for us but maybe a consideration for others?
2. Most build systems really don't like file names that aren't known in advance. You can do it, it's just more of a pain. I guess you could do it where you specify the number of files to use and it *always* generates `model_0.c` .. `model_{n-1}.c`, etc. That would be ok.
3. It will make `#include`ing the code in another file more awkward. I think @arichardson and me both do this. And I'm planning to try to wrap the whole file in `class Model { ... };` so that it doesn't use any global state and I can instantiate more than one of them; I haven't tried that yet though.

> M1 MacBook Pro

Ha that would be why it's so much faster for you than me!

We also are already using OCaml 5 so no issue for us there. I think installing OCaml 5 is exactly as painful as installing OCaml 4 so I'd say requiring people to upgrade is perfectly reasonable.

Thanks so much for working on this!

Alasdair, 2023-10-27:
I pushed another commit that has the further improvements I was working on, and has the heuristic for the constant folding I mentioned.

arichardson, 2023-10-27:
> I pushed another commit that has the further improvements I was working on, and has the heuristic for the constant folding I mentioned.

Would it be possible to make a new release of sail so that a plain opam install pulls in the fixed version? Might also be an argument to bump the required sail version.

Alasdair, 2023-10-27:
Yes, I should be able to do that next week

Alasdair, 2023-10-31:
I've submitted a new release to opam so should be available whenever it gets merged in. Got things down to about 10 seconds and 300mb of memory usage so that's probably enough for now.

billmcspadden-riscv, 2023-10-31:
Another datapoint on compilation performance:

I'm running on  a dedicated Linux server (an old Dell desktop: 4 proc,  intel i5,  16GB memory, 2TB SSD),  and the
performance change for me is small: less than 4 minutes wall clock time.

rsnikhil, 2023-11-11:
> I tried again with the latest `sail2`.

What is `sail2`?

I am astonished at the short times cited in this issue-thread.  When I do a vanilla installs per the READMEs (opam, ocaml, sail), compiling the sail-riscv model (`make csim`) takes 37 minutes on a Mac M2 (Ventura 13.6, 24 GB mem, 2TB disk).

jrtc27, 2023-11-11:
> What is sail2?

The branch; sail2 contains the current version of the language, different to the old original sail.

rsnikhil, 2023-11-11:
> > What is sail2?
> 
> The branch; sail2 contains the current version of the language, different to the old original sail.

How do I get sail2?  ‘opam list sail2’ says there’s no match.  Do I need to build it from scratch in the repo?
Are all the above cited model-build times for sail2 (i.e., is that the reason, you think, for my 37-minute build time)?

Alasdair, 2023-11-11:
We have instructions here https://github.com/rems-project/sail/blob/sail2/INSTALL.md

For historical reasons sail2 is just he equivalent of the master/main branch in our repository

rsnikhil, 2023-11-12:
Ok, problem solved.

I was able to install sail2 (thanks to @Alasdair   for pointers and help!).

Building the sail-riscv model ('make clean csim') now takes 1 min 23 secs (compared to 37 secs with sail 0.16) on a Mac M2 (Ventura 13.6, 24 GB mem, 2TB disk)

Alasdair, 2023-11-13:
I don't think so. I think things that were using the `master` branch are now using the `legacy` branch, which is for some models we wrote a long time ago and never kept up to date.

Timmmm, 2023-11-15:
I tried again and I'm still not seeing compile times anywhere near as low as you guys are reporting or what I was getting before vector, though memory use is much lower than it used to be.

* sail-riscv: 9f4cad61fe4713a46f9586e23c5f60dec79aece6
* sail: 3f8a32030253291d391e915d6b0bedd628536398
* GCC: 8.5.0

Command: `ARCH=RV64 make c_emulator/riscv_sim_RV64`

Wall times/max RSS:

* Sail compilation: 1m 8s; 545 MB
* C compilation: 6m 17s; 1147 MB
* Total: 7m 35s

Also some people mentioned issues with Clang, but I don't see how that can be given that the sail-riscv Makefile hard-codes GCC.

I don't think my laptop is *that* bad, so I'd say there's still some improvement to be done. I guess it's possible that more recent GCC versions improved things. I'll try that. I'll also try on a different machine.

Alasdair, 2023-11-15:
gcc is a symlink to clang on macOS by default. On and M1 Pro with 32GB RAM I get the following:

- Sail compilation:
```
real 17.00
user 15.16
sys 1.37
           583139328  maximum resident set size
        151282692987  instructions retired
         40959533901  cycles elapsed
           522340352  peak memory footprint
```
- C compilation
```
real 14.11
user 13.77
sys 0.37
           840384512  maximum resident set size
           109460937  instructions retired
            52061459  cycles elapsed
             2606144  peak memory footprint
```
Which seems ballpark-similar in terms of memory usage to what you get but a lot faster. I'll try using my Linux desktop in a bit. What kind of specs does your laptop have? If it has a mechanical hard disk (or even a slower SSD) and temporary files aren't going to some kind of memory-backend tmpfs that could slow things down a lot. The fact that the C compilation is also proportionally a lot slower makes me think it isn't purely a Sail issue.

Alasdair, 2023-11-15:
An on Linux (Ryzen 2600X, 16GB RAM):
```
Sail: 19.19user 1.42system 0:20.65elapsed 99%CPU (0avgtext+0avgdata 552012maxresident)k
C: 54.85user 1.46system 0:56.52elapsed 99%CPU (0avgtext+0avgdata 618584maxresident)k
```

bacam, 2023-11-15:
Note that timings for Sail depend on whether you have a recent `z3_problems` cache (which `make clean` will remove).  For reference, on an `i7-10710U` I got 2m 56s for the sail step without a cache, 37s with a cache, and the gcc step took 1m 24s.  The max RSS was under 1GB for both.  This was on Ubuntu 22.04 with gcc 11.4.0.

Alasdair, 2023-11-15:
I usually use `make -B` rather than `make clean` when I want to rebuild everything to avoid that

Timmmm, 2023-11-16:
> gcc is a symlink to clang on macOS by default

Ah of course.

I have some Dell laptop with an i5-1245U. Passmark gives it 3150 single thread compared to 3812 for the M1 Pro. I'll try with some different machines & compilers.

Alasdair, 2023-11-16:
Is it native linux or WSL?

Timmmm, 2023-11-16:
Native RHEL 8.

jrtc27, 2023-11-16:
Have you tried a different compiler version (or compiler in general) to see if it's your GCC or your system?

Timmmm, 2023-11-16:
Not yet but I will do.

Alasdair, 2023-11-16:
Hmm, I'm not sure what could be causing it then. The processor you have should be faster than my desktop. You could try using 'perf' to get some more detailed information, but I'm not sure which options you'd want to use.

Timmmm, 2024-04-29:
I'm still seeing this but it's not urgent because I have vector commented out for now. I did make a Dockerfile to try to reproduce it and tried it on a different machine, which showed a slow down but nowhere near what I see on my laptop.

I'll close this and look into it again when I need vector.

closed 2024-04-29 by Timmmm