riscv/sail-riscv #325: Fail to run 'make' for sail-riscv

xiaomaaaaa, 2023-10-25:
I'm on `sail -v` = `Sail 0.16 (sail @ opam-v2.1.5)`
When trying to run 'make' for sail-riscv on Centos7 I get the following error:
`Warning: Incomplete pattern match statement at model/riscv_insts_vext_fp.sail:722.77-82:
722 |                              let (fflags, elem) : (bits_fflags, bits('m)) = match 'm {
    |                                                                             ^---^
    | 
The following expression is unmatched: 64

Warning: Incomplete pattern match statement at model/riscv_insts_vext_fp.sail:731.77-82:
731 |                              let (fflags, elem) : (bits_fflags, bits('m)) = match 'm {
    |                                                                             ^---^
    | 
The following expression is unmatched: 64

Warning: Incomplete pattern match statement at model/riscv_insts_vext_red.sail:213.12-17:
213 |      sum = match funct6 {
    |            ^---^
    | 
The following expression is unmatched: FVV_VFWREDOSUM

echo "declare {isabelle} rename field sync_exception_ext = sync_exception_ext_exception" >> generated_definitions/lem/RV64/riscv_types.lem
lem -isa -outdir generated_definitions/isabelle/RV64 -lib Sail=/home/mashixiong0318004377/.opam/5.1.0/share/sail/src/lem_interp -lib Sail=/home/mashixiong0318004377/.opam/5.1.0/share/sail/src/gen_lib \
	handwritten_support/0.11/riscv_extras.lem handwritten_support/0.11/mem_metadata.lem handwritten_support/0.11/riscv_extras_fdext.lem \
	generated_definitions/lem/RV64/riscv_types.lem \
	generated_definitions/lem/RV64/riscv.lem
File "generated_definitions/lem/RV64/riscv.lem", line 24483, character 7:
  Syntax error
make: *** [Makefile:330: generated_definitions/isabelle/RV64/Riscv.thy] Error 1
`
and the content of the ‘riscv.lem’ is as follows:
`let riscv_f16Rsqrte7 rm v =
   let class = (fp_class v  : mword ty16) in      #line 24483
   let b__0 = class in
   if ((eq_vec b__0 (0x0001 : mword ty16))) then
     return ((nvFlag ()  : mword ty5), (0x7E00 : mword ty16))
   else if ((eq_vec b__0 (0x0002 : mword ty16))) then
     return ((nvFlag ()  : mword ty5), (0x7E00 : mword ty16))
   else if ((eq_vec b__0 (0x0004 : mword ty16))) then
     return ((nvFlag ()  : mword ty5), (0x7E00 : mword ty16))
   else if ((eq_vec b__0 (0x0100 : mword ty16))) then
     return ((nvFlag ()  : mword ty5), (0x7E00 : mword ty16))
   else if ((eq_vec b__0 (0x0200 : mword ty16))) then
     return ((zeros_implicit (5:ii)  : mword ty5), (0x7E00 : mword ty16))
   else if ((eq_vec b__0 (0x0008 : mword ty16))) then
     return ((dzFlag ()  : mword ty5), (0xFC00 : mword ty16))
   else if ((eq_vec b__0 (0x0010 : mword ty16))) then
     return ((dzFlag ()  : mword ty5), (0x7C00 : mword ty16))
   else if ((eq_vec b__0 (0x0080 : mword ty16))) then
     return ((zeros_implicit (5:ii)  : mword ty5), (0x0000 : mword ty16))
   else if ((eq_vec b__0 (0x0020 : mword ty16))) then
     (rsqrt7 v true  : M (mword ty64)) >>= fun (w__0 : mword ty64) ->
     return ((zeros_implicit (5:ii)  : mword ty5),
             (subrange_vec_dec w__0 (15:ii) (0:ii)  : mword ty16))
   else
     (rsqrt7 v false  : M (mword ty64)) >>= fun (w__1 : mword ty64) ->
     return ((zeros_implicit (5:ii)  : mword ty5),
`
Any advices would be very helpful for me.Thank you!

Alasdair, 2023-10-26:
Can you do `make csim` or even just `make check` (maybe after doing `make clean`)?

I can't reproduce these issues with the Lem -> Isabelle path. I think in general just `make` is trying to build too much which isn't useful to most users. I'll see about making a PR to simplify what the Makefile is trying to do by default.

xiaomaaaaa, 2023-10-30:
I can do `make csim `  `make osim`  `make rvfi` , but when I do `make riscv_isa` , The above problem arises。

ingmarfjolla, 2023-11-17:
adding to this one since my issue was a duplicate of this, changing the sail-riscv commit to the one mentioned in issue: #333 https://github.com/riscv/sail-riscv/commit/532714a6c71b47a91176eb90fef3b3b049c52fce ended up fixing the issue. 

I tested on both ocaml version 4.13.1 (which I think the CI here uses) and ocaml 5.1.0 with sail version 0.17.1 and the make file ran. I also tested with the ocaml/opam:latest image that this repos dockerfile uses and the ubuntu:latest image and it works on that commit for now.

Alasdair, 2023-11-18:
Should be fixed by https://github.com/riscv/sail-riscv/pull/353

The very large bitvectors in the vector extension cause issues for the mapping into Isabelle and HOL4 machine word libraries, so while that commit fixes the build it might make the generated theorem prover definitions less useful in the short term until we can fix the underlying issues.

closed 2023-12-19 by billmcspadden-riscv