riscv/sail-riscv #337: Encoding and assembly of `vsetvl` are incorrect

ThinkOpenly, 2023-10-31:
In commit c90cf2e6eff5f, new file `model/riscv_insts_vext_vset.sail` was added and includes definitions for instructions `vsetvli` and `vsetvl`. Therein, the encoding for `vsetvl` is define thus:
```
encdec_vsetop(op) @ ma @ ta @ sew @ lmul @ rs1 @ 0b111 @ rd @ 0b1010111
```
where `encdec_vsetop(VSETVL) = 0b1000`

Per my reading of the "RISC-V "V" Vector Extension" specification, Version 1.0, the encoding for `vsetvl` (p. 25), the encoding should look more like:
```
0b1000000 @ rs2 @ rs1 @ 0b111 @ rd @ 0b1010111
```

Similarly, the assembly for `vsetvl` is defined in Sail as:
```
vsettype_mnemonic(op) ^ spc() ^ reg_name(rd) ^ sep() ^ reg_name(rs1) ^ sep() ^ sew_flag(sew) ^ maybe_lmul_flag(lmul) ^ maybe_ta_flag(ta) ^ maybe_ma_flag(ma)
```

But my reading of the spec shows something more like:
```
vsettype_mnemonic(op) ^ spc() ^ reg_name(rd) ^ sep() ^ reg_name(rs1) ^ sep() ^ reg_name(rs2)
```

Lastly, the discriminator to mnemonic mapping in Sail is:
```
mapping vsettype_mnemonic : vsetop <-> string ={
  VSETVLI <-> "vsetvli",
  VSETVL  <-> "vsetvli"
}
```

But, the `VSETVL` mnemonic should obviously be `vsetvl`.

ThinkOpenly, 2023-11-09:
@XinlaiWan, FYI

XinlaiWan, 2023-11-22:
Thanks! I've modified the incorrect code.

PR #359 (2024-05-16): Fix the encoding and assembly of `vsetvl` instruction
The implementation of `vsetvl` and `vsetvli` are rearranged, and some helper functions are added.

This PR fixed #337 .
commit 9692f84f (2024-05-16): Rewrite the implementation of vsetvli and vsetvl instructions to fix #337

The issue shows that the encoding and assembly definitions of vsetvl and vsetvli are incorrectly mixed together,
but they actually differ a lot and therefore cannot have a unified definition as the old implementation did,
so this section has been rewritten to split their definitions and executions.
And because these instructions have similar functional behavior after the decoding stage,
some common code is further extracted to be helper functions in this commit.