riscv/sail-riscv #338: rs1 and rs2 operands are in the wrong order for store conditional assembly

Timmmm, 2023-11-06:
```
mapping clause assembly = STORECON(aq, rl, rs2, rs1, size, rd)
  <-> "sc." ^ size_mnemonic(size) ^ maybe_aq(aq) ^ maybe_rl(rl) ^ spc() ^ reg_name(rd) ^ sep() ^ reg_name(rs1) ^ sep() ^ reg_name(rs2)
```

Should be

```
mapping clause assembly = STORECON(aq, rl, rs2, rs1, size, rd)
  <-> "sc." ^ size_mnemonic(size) ^ maybe_aq(aq) ^ maybe_rl(rl) ^ spc() ^ reg_name(rd) ^ sep() ^ reg_name(rs2) ^ sep() ^ reg_name(rs1)
```

`rs2` is first. Is there actually a specification for how RISC-V assembly works anywhere? The only way I was really able to confirm this was from reverse engineering the example code in the spec:

[image]

jrtc27, 2023-11-06:
If you want it to be valid assembly you need the parentheses to make it a memory operand. And no, there isn't a comprehensive specification for what the de-facto assembly syntax is.

Timmmm, 2023-11-06:
Good point! So

```
mapping clause assembly = STORECON(aq, rl, rs2, rs1, size, rd)
  <-> "sc." ^ size_mnemonic(size) ^ maybe_aq(aq) ^ maybe_rl(rl) ^ spc() ^ reg_name(rd) ^ sep() ^ reg_name(rs2) ^ sep() ^ "(" ^ reg_name(rs1) ^ ")"
```

ThinkOpenly, 2023-11-13:
> ```
> mapping clause assembly = STORECON(aq, rl, rs2, rs1, size, rd)
>   <-> "sc." ^ size_mnemonic(size) ^ maybe_aq(aq) ^ maybe_rl(rl) ^ spc() ^ reg_name(rd) ^ sep() ^ reg_name(rs2) ^ sep() ^ "(" ^ reg_name(rs1) ^ ")"
> ```

I agree. I just stumbled on this, too. There are some bad examples with incorrect syntax in section A.3.3 of the RISC-V Instruction Set Manual (20191213).

PR #345 (2023-11-29): Fix Store-Conditional assembly operand order and add parens
The operand order for Store-Conditional assembly has the second and third operands reversed.

The RISC-V Instruction Set Manual states:
> ```
> SC.W conditionally writes a word in rs2 to the address in rs1 [...]
> If the SC.W succeeds, the instruction writes the word in rs2 to memory,
> and it writes zero to rd. If the SC.W fails, the instruction does not
> write to memory, and it writes a nonzero value to rd.
> ```

`rd` is for the return code, `rs2` is the value, and `rs1` is the memory address.

For the syntax `sc.w A,B,(C)`:
- `A` [... cut]