riscv/sail-riscv #343: sail-riscv build fails

DenizzzGuzell, 2023-11-13:
Hi,
I am trying to install sail-riscv. I installed sail using opam, but i get the following error when i run make inside sail-riscv folder.

[image]

None of the make commands are not working. I tried to reinstall package using `opam reinstall sail -y `. It still tell me that there is syntax error in one of the model file named `prelude.sail` . I am currently working on Ubuntu Server 5.4.0.

Opam version is 2.1.5
Z3  version is 4.4.1
Sail version is 0.14.0

Detailed packages installed in opam can be seen from picture below.

[image]

I have followed RISCOF user guide section 3.5 named Quickstart to build sail_riscv. Commands that I had tried to use are :
`$ sudo apt-get install opam  build-essential libgmp-dev z3 pkg-config zlib1g-dev`

`$ opam init -y --disable-sandboxing`

`$ opam switch create ocaml-base-compiler.4.06.1`

`$ opam install sail -y`

`$ eval $(opam config env)`

`$ git clone https://github.com/riscv/sail-riscv.git`

`$ cd sail-riscv`

`$ make //Error occured`

`$ ARCH=RV32 make`

`$ ARCH=RV64 make`

`$ ln -s sail-riscv/c_emulator/riscv_sim_RV64 /usr/bin/riscv_sim_RV64`

`$ ln -s sail-riscv/c_emulator/riscv_sim_RV32 /usr/bin/riscv_sim_RV32`

It would be great if you can help me. Thank you.

jrtc27, 2023-11-13:
Sail 0.14 is ancient, use the latest version (0.16?)

jrtc27, 2023-11-13:
Per https://github.com/rems-project/sail/blob/sail2/INSTALL.md this is because the OCaml version you selected is too old.

jrtc27, 2023-11-13:
> Ubuntu Server 5.4.0

FYI this is not a thing. Ubuntu versions are of the form YY.MM(.n). This is likely the Linux kernel version in use instead; Ubuntu 20.04 LTS (“focal”) uses that kernel version.

closed 2023-11-13 by jrtc27

Alasdair, 2023-11-13:
`opam switch create 5.1.0`
followed by
`eval $(opam env)`
should get you a newer version of OCaml that will work.