riscv/sail-riscv #346: HTIF error

MihailBeloshapkin, 2023-11-14:
Hi, I am currently trying to run simple RISC-V program with Sail emulator. I compiled it with cross compiler and trying to run it on an OCaml emulator. And I get the following error:

`CSR mstatus <- 0x0000000A00001800`
`htif[0x0000000000000000] -> 0x0000000000000000`
`trapping from M to M to handle fetch-access-fault`
`handling exc#0x01 at priv M with tval 0x0000000000000000`
`reservation <- none
`
As well as i know HTIF is an interface that allows to access peripheral devices. I didn't find a guide on how to set it up. So how can I fix this problem?

billmcspadden-riscv, 2023-11-14:
Hi Mikhail.

Is there a reason you are using the OCAML emulator rather than the C
emulator?
Most people are using the C emulator.  For me to help debug this,  it would
be
helpful for you to run the C emulator.

For simple tests, I only use the htif interface to indicate that my code
has completed.
It appears you might be using it for other purposes, yes?

Bill Mc.

MihailBeloshapkin, 2023-11-15:
I need to use OCaml emulator because of my university diploma project (researching methods to speed up Sail RISC-V emulation and implementing them into Sail -> OCaml compiler). So first of all it is necessary for me to understand how to set up Sail and its environment.

For the first example I use simplest C program (main function that returns 0, without stdlib inclusion)
Compiled:
`riscv64-unknown-elf-gcc hello.c -o hello`

And for this command (if I use C emulator):
`./c_emulator/riscv_sim_RV64 ../../../../riscv/hello`

I get this output:
`Running file ../../../../riscv/hello.`
`ELF Entry @ 0x10116`
`Unable to locate htif tohost port.`

jrtc27, 2023-11-15:
Declaring two uint64_t (non-static visibility) variables tohost and fromhost should be sufficient for the Sail model.

MihailBeloshapkin, 2023-11-18:
I changed source code:
``````
```CSR mstatus <- 0x0000000A00001800
within_phys_mem: 0x0000000000000000 not within phys-mem:
  plat_rom_base: 0x0000000000001000
  plat_rom_size: 0x0000000000001000
  plat_ram_base: 0x0000000080000000
  plat_ram_size: 0x0000000004000000
trapping from M to M to handle fetch-access-fault
handling exc#0x01 at priv M with tval 0x0000000000000000
```

MihailBeloshapkin, 2023-11-18:
Compiler installed from here: https://github.com/riscv/riscv-gnu-toolchain

MihailBeloshapkin, 2023-11-18:
I changed source code:

volatile unsigned long tohost;
volatile unsigned long fromhost;
```
int main()
{
  return 0;
}
```
and now this message is displayed in an infinite loop
```
within_phys_mem: 0x0000000000000000 not within phys-mem:
  plat_rom_base: 0x0000000000001000
  plat_rom_size: 0x0000000000001000
  plat_ram_base: 0x0000000080000000
  plat_ram_size: 0x0000000004000000
trapping from M to M to handle fetch-access-fault
handling exc#0x01 at priv M with tval 0x0000000000000000```

jrtc27, 2023-11-19:
Sounds like your code is branching to address 0? What are you using for the program entry point / as your libc and C start-up code?

Alasdair, 2023-11-20:
```
volatile unsigned long tohost;
volatile unsigned long fromhost;

int main()
{
  return 0;
}
```
If you are running a C program on what is essentially a bare metal target, there are a few things you will need to consider:

- First, how does the processor know to start executing a function called main in the first place?
- Second, what does return do? There is no operating system to return to!

To solve the first you'll likely need some kind of linker script that says where your code will live in physical memory. You will want to place some code at the location of the program counter when the CPU starts so that code starts executing. Likely you'll want to write this code in assembly, then it can call your C main function. For the second you'll either want you program to either just spin in a loop or I think there is a special address you can write to to terminate the simulator.

A while ago I wrote some C code like this to run on our ARM model: https://github.com/Alasdair/sail-arm-boot/tree/master, you could take a look at that for some inspiration.

jordancarlin, 2026-11-09:
Closing since this was a question about a specific program and the OCaml emulator no longer exists anyway.

closed 2026-11-09 by jordancarlin