riscv/sail-riscv #354: Help with compile error

ved-rivos, 2023-11-19:
I am hitting this "Type error" but there is no indication about what its complaining about. Please help.

```
mkdir -p generated_definitions/ocaml/RV64
sail -dno_cast -ocaml -ocaml-nobuild -ocaml_build_dir generated_definitions/ocaml/RV64 -o riscv model/prelude.sail model/prelude_mapping.sail model/riscv_xlen64.sail model/riscv_flen_D.sail model/riscv_vlen.sail model/prelude_mem_metadata.sail model/prelude_mem.sail model/riscv_types_common.sail model/riscv_types_ext.sail model/riscv_types.sail model/riscv_vmem_types.sail model/riscv_reg_type.sail model/riscv_freg_type.sail model/riscv_regs.sail model/riscv_pc_access.sail model/riscv_sys_regs.sail model/riscv_pmp_regs.sail model/riscv_pmp_control.sail model/riscv_ext_regs.sail model/riscv_addr_checks_common.sail model/riscv_addr_checks.sail model/riscv_misa_ext.sail model/riscv_vreg_type.sail model/riscv_vext_regs.sail model/riscv_csr_map.sail model/riscv_vext_control.sail model/riscv_next_regs.sail model/riscv_sys_exceptions.sail model/riscv_sync_exception.sail model/riscv_next_control.sail model/riscv_softfloat_interface.sail model/riscv_fdext_regs.sail model/riscv_fdext_control.sail model/riscv_csr_ext.sail model/riscv_sys_control.sail model/riscv_platform.sail model/riscv_mem.sail model/riscv_pte.sail model/riscv_ptw.sail model/riscv_vmem_common.sail model/riscv_vmem_tlb.sail model/riscv_vmem_sv39.sail model/riscv_vmem_sv48.sail model/riscv_vmem_rv64.sail model/riscv_types_kext.sail model/riscv_insts_begin.sail model/riscv_insts_base.sail model/riscv_insts_aext.sail model/riscv_insts_cext.sail model/riscv_insts_mext.sail model/riscv_insts_zicsr.sail model/riscv_insts_next.sail model/riscv_insts_hints.sail model/riscv_insts_fext.sail model/riscv_insts_cfext.sail model/riscv_insts_dext.sail model/riscv_insts_cdext.sail model/riscv_insts_zba.sail model/riscv_insts_zbb.sail model/riscv_insts_zbc.sail model/riscv_insts_zbs.sail model/riscv_insts_zfh.sail model/riscv_insts_zfa.sail model/riscv_insts_zkn.sail model/riscv_insts_zks.sail model/riscv_insts_zbkb.sail model/riscv_insts_zbkx.sail model/riscv_insts_zicond.sail model/riscv_zicfilp_regs.sail model/riscv_insts_zicfilp.sail model/riscv_insts_vext_utils.sail model/riscv_insts_vext_vset.sail model/riscv_insts_vext_arith.sail model/riscv_insts_vext_fp.sail model/riscv_insts_vext_mem.sail model/riscv_insts_vext_mask.sail model/riscv_insts_vext_vm.sail model/riscv_insts_vext_red.sail model/riscv_jalr_seq.sail model/riscv_insts_end.sail model/riscv_step_common.sail model/riscv_step_ext.sail model/riscv_decode_ext.sail model/riscv_fetch.sail model/riscv_step.sail model/riscv_analysis.sail
Warning: Deprecated model/prelude.sail:80.22-43:
80 |val string_startswith = "string_startswith" : (string, string) -> bool
   |                      ^-------------------^
   | 
All external bindings should be marked as either monadic or pure

Warning: Deprecated model/riscv_vlen.sail:41.36-49:
41 |val get_elen_pow : unit -> {|5, 6|} effect {rreg}
   |                                    ^-----------^
   | 
Explicit effect annotations are deprecated. They are no longer used and can be removed.

Warning: Duplicate function type definition for spc_forwards
model/prelude.sail:306.4-16:
306 |val spc_forwards : unit -> string
    |    ^----------^
    | This duplicate definition is being ignored!
Type error:
No overloading for vector_subrange, tried:
* subrange_bits
   Could not resolve quantifiers for subrange_bits
   * (0 <= 63 & (63 <= 63 & 63 < 63))
make: *** [Makefile:226: generated_definitions/ocaml/RV64/riscv.ml] Error 1

```

closed 2023-11-19 by ved-rivos

Alasdair, 2023-11-20:
Errors without correct location information are bugs I would like to fix. If you could provide concrete steps to reproduce I would appreciate it.

ved-rivos, 2023-11-20:
@Alasdair - this can be reproduced with the following change:
```
diff --git a/model/riscv_sys_regs.sail b/model/riscv_sys_regs.sail
index f472ca2..b155114 100644
--- a/model/riscv_sys_regs.sail
+++ b/model/riscv_sys_regs.sail
@@ -221,8 +221,8 @@ bitfield Mstatush : bits(32) = {
 }
 register mstatush : Mstatush
 
-bitfield Mstatus : xlenbits = {
-  SD   : xlen - 1,
+bitfield Mstatus : bits(63) = {
+  SD   : 63,

```