riscv/sail-riscv #363: Partial PMP match reports original address in tval

defermelowie, 2023-11-29:
From the priv. spec.:

> If mtval is written with a nonzero value when a misaligned load or store causes an access-fault or page-fault exception, then mtval will contain the virtual address of the portion of the access that caused the fault.

So, if an misaligned load spans 2 PMP regions, the first with sufficient access and the second without, a load-access-fault should be triggered and  the value in `tval` should be the start of the second PMP region? 

This is not the case in this model since the [`process_load`](https://github.com/riscv/sail-riscv/blob/b414baefb648cfef59d3cebcd457fdaadf6550ed/model/riscv_insts_base.sail#L375C72-L375C72) function puts the original address in `sync_exception.excinfo` (via [`handle_mem_exception`](https://github.com/riscv/sail-riscv/blob/153f983e25c8789b92b4359e2c6c4626a2fef023/model/riscv_sys_control.sail#L534C10-L534C30)) which is later used to set `mtval` & `stval`.

defermelowie, 2023-11-30:
The relevant part of [the emulator log](https://github.com/riscv/sail-riscv/files/13511141/misaligned.sail.log) is shown below:

```log
mem[X,0x00000000800001DA] -> 0x4185
[113] [S]: 0x00000000800001DA (0x4185) c.li gp, 1
x3 <- 0x0000000000000001
mem[X,0x00000000800001DC] -> 0x6283
mem[X,0x00000000800001DE] -> 0x0005
[114] [S]: 0x00000000800001DC (0x00056283) lwu t0, 0(a0)
mem[R,0x0000000080002000] -> 0x55555555
x5 <- 0x0000000055555555
mem[X,0x00000000800001E0] -> 0x4189
[115] [S]: 0x00000000800001E0 (0x4189) c.li gp, 2
x3 <- 0x0000000000000002
mem[X,0x00000000800001E2] -> 0x6283
mem[X,0x00000000800001E4] -> 0x0025
[116] [S]: 0x00000000800001E2 (0x00256283) lwu t0, 2(a0)
trapping from S to M to handle load-access-fault
handling exc#0x05 at priv M with tval 0x0000000080002002
CSR mstatus <- 0x0000000A00000880
```

allenjbaum, 2023-12-02:
I think the exception logic for misaligned accesses is probably far more
broken than this.
I've tried to categorize the possible responses to a misaligned access,
which depends on hidden microarchitecural level details:
 - does it support misaligned loads
     (It isn't clear if this property is must be deterministic, i.e. in the
absences of any other exception,
      will a compatible implementation never trap with a misaligned cause
(to DRAM),
      or might it sometimes trap depending on some microarchitectural
state.)
 - does it access the lower address first, or the higher address? Does that
depend on the BigEndian property?
 - does it implement misaligned trap priority higher or lower than
page/access fault (I am assuming this is a static property, but the spec
doesn't mandate that)
 - is the access split into 2 separate accesses or treated as a single
access (

Depending on those, and whether the either half would trap if accessed on
its own ("neither" in the "low-first" column),
this is what I think happens:

HW misalign hipri-misalign low-first high/low legal? CAUSE TVAL mem
0 1 x x   misalign low unchg
0 0 1 errH,OK   errH low unchg
0 0 1 OK, errL errL low unchg
0 0 1 errH,errL   errL low unchg
0 0 0 errH,OK   errH hi unchg
0 0 0 OK, errL errL hi unchg
0 0 0 errH,errL   errH hi unchg
1 x x errH,OK   errH hi lo
1 x x OK, errL errL low hi
1 x 1 errH,errL errL low unchg
1 x 0 errH,errL errH hi unchg
0 x neither OK, OK   -- -- hi,lo
0 x neither OK, OK -- -- hi,lo
1 x neither OK, OK -- -- hi,lo
1 x neither OK, OK   -- -- hi,lo
There may be more unpredictability than this, for example if the access is
split into high and low, e.g.
- does the implementation wait until both accesses determine whether they
trap or not and choose the highest priority if both would trap?
Or does it take them in high/low or low/high order? (thisi chart assume the
latter, but the spec is silent)
- is high/low ordering static, or can it change dynamically