riscv/sail-riscv #374: Suspicious dead code in vfncvt instructions

bacam, 2023-12-12:
While experimenting with strengthening Sail's type system to rule out some impossible cases, I noticed that the `SEW = 'm = 8` cases in the `VFNUNARY0` execution function are unreachable.  They are ruled out by the `illegal_fp_variable_width` check, which appears to treat the instructions as if they were widening operations rather than narrowing operations.  Perhaps this check was supposed to be a little different?

aamartin0000, 2023-12-12:
perhaps it's a placeholder, for eventual inclusion of FP8?

bacam, 2023-12-13:
I don't think so: there are both integer and float cases, and the float cases are explicit errors.

Alasdair, 2023-12-13:
It looks suspect to me, but I wasn't able to understand the wording of the vector spec well enough to really understand what it should be doing.

allenjbaum, 2023-12-13:
vfncvt is explicitly narrowing (that's what the N in vfNcvt is for) from a
double width (64b) source to a singlewidth (32b) result, with at least one
operand being a float
  64b                float-> 32b [un]signed int [truncate]
  64b                float-> 32b float [round to odd]
  64b [un]signed int-> 32b float

I am unaware of any FP8 or even FP16 plans that would change that...yet.
Neither format is part of the existing spec, though I expect that to change
But if there were any, I'd expect FP32->FP16, and FP16->FP8 .

Alasdair, 2023-12-13:
The current code is implementing `32 -> 16`, and `16 -> 8` conversions, which are being ruled out by the current legality checks. If they are not part of the spec we should remove them I guess?

Alasdair, 2023-12-13:
Doing the same legality check as the widening operation still seems suspect to me, as the spec states that in one case SEW, LMUL, etc refer to the destination, and in the other they refer to the source. That's the bit I don't fully understand.