riscv/sail-riscv #378: 32-bit vs. 34-bit physical addresses in RV32 (without/with Sv32)

rsnikhil, 2023-12-18:
'mem_read_priv()' [riscv_mem.sail, L180] declares the address argument as 'xlenbits'

But Sv32 phys mem addrs are 34 bits, not 32 (XLEN) bits.

If RV32 is configured WITHOUT Sv32, are phys addrs 32 bits or 34 bits?

If RV32 is configured WITH Sv32, but we are running at M (machine) privilege or with SBare vmem, we can only generate 32-bit addresses.  Are these simply zero-extended to 34 bits?

allenjbaum, 2023-12-18:
This is more of an ISA question than a Sail question.
Without Smode, there is no way to generate more than 32 bits of address, so
that's a requirement.
With Smode (and SV32 enabled) OR (Mmode && mstatus.MPRV=1  && mstatus.PP !=
Mmode && SATP.mode !=bare) you can generate a 34 bit address.
RISC-V physical addresses are always zero extended (VAs are sign extended)
so the top 2 bits would be zero otherwise

rsnikhil, 2024-01-02:
The question is, what should the Sail model do?

I think the current definition of physical addrs as 32 bits for RV32 is not correct.  With the 32 bit setting, the Sail model cannot model implementations that populate memory above the 32-bit address mark.

allenjbaum, 2024-01-03:
It's worse than that.
Without VM enabled, you can only use 34 bit addressing in Mmode by setting
MPRV and giving it a VA that translates to the PA you want (and taking all
the precautions to ensure that traps aren't fatal)
SATP has to point to a legal page table under those conditions; it can't be
in bare mode (because in bare mode, the rest of SATP must be zero)
and if I'm reading the spec 3.1.6.2 correctly, even then there are cases
where Mmode can only access half of the 32b address space
(specifically, in RV64, in MMode with SXL=32, all register operands are
sign extended, not zero extended, and that would apply to address
calculations)

rsnikhil, 2024-01-03:
@allanjbaum I think you're discussing different points than the issue I am raising.

You're discussing the various ISA nuances of Sv32 translation (when it occurs, how it occurs, …), which his fine.

My point is is different, it’s about what happens to the final _output_ of Sv32 translation: that is a 34-bit physical address, and the sail-riscv model currently assumes it is 32 bit.

allenjbaum, 2024-01-04:
Oh, yes, I was just pointing out that the spec has some corner cases that I
don't think have been thought through (e.g. are PAs sign extended or zero
extended?) that are related.
It gets even worse if XLEN is different between Mode and U/S mode, which
Sail can't handle at all.

KotorinMinami, 2024-12-19:
The #641 has now been merged, and 34-bit physical addresses are now supported under RV32. Although there is currently a lack of test, this issue can still be addressed. In Sbare mode, we actually zero-extend the 32-bit physical address to 34 bits

Timmmm, 2024-12-19:
Yeah I think let's keep this open until we have a test for it. I might write one.

closed 2025-09-16 by Timmmm