riscv/sail-riscv #399: Make file error

Ludini1, 2024-02-02:
When running make, I get the following error:

`File "generated_definitions/lem/RV64/riscv.lem", line 346, character 35 to line 346, character 48
  Type error: unbound variable: parse_hex_bits
make: *** [Makefile:327: generated_definitions/isabelle/RV64/Riscv.thy] Error 1`

I'm using WSL so I already installed the latest development versions of Z3 and Sail but it resulted in the same error anyway.

bacam, 2024-02-02:
The theorem prover backends have fallen a little behind and don't yet have `parse_hex_bits`.  If you don't need Isabelle output I suggest just making the target you want, probably `c_emulator/riscv_sim_RV64` or `c_emulator/riscv_sim_RV32`.  (It would be nice if these were the default, but unfortunately #329 appears to have stalled.)

shareefj, 2024-02-08:
Just for anyone else flailing around like I was, the correct make command seems to be:

```
ARCH=32 make c_emulator/riscv_sim_RV32
```

or s/32/64/g if that's your bag.

Incarnation-p-lee, 2024-03-25:
Just FYI, still have this issue in:
```
[NOTE] Package sail is already installed (current version is 0.17.1).
```

shareefj, 2024-03-25:
> Just FYI, still have this issue in:
> 
> ```
> [NOTE] Package sail is already installed (current version is 0.17.1).
> ```

If you uninstall the package, then re-install it, that message will go away.  ;-)

Incarnation-p-lee, 2024-03-26:
Take below step(s), not sure if I missed something or you don't mean the error message.
```
opam remove sail
opam clean
opam install sail -y
  ⬇ retrieved sail.0.17.1  (https://opam.ocaml.org/cache)
  ∗ installed sail.0.17.1 
  Done.
ARCH=RV64 make
```
then still have the error message as below
```
File "generated_definitions/lem/RV64/riscv.lem", line 346, character 35 to line 346, character 48
  Type error: unbound variable: parse_hex_bits
make: *** [Makefile:329: generated_definitions/isabelle/RV64/Riscv.thy] Error 1
```

Alasdair, 2024-03-26:
I will look into making a PR to fix this later this week. I still think in general the Makefile is trying to build too many targets by default including some targets that are inherently a little brittle.

Incarnation-p-lee, 2024-03-26:
> I will look into making a PR to fix this later this week. I still think in general the Makefile is trying to build too many targets by default including some targets that are inherently a little brittle.

That's Great!

Alasdair, 2024-04-03:
This specific error is fixed in Sail by:

https://github.com/rems-project/sail/commit/297e4d270acc6d273d27ad80cb3bf0ef78f7f985

I should probably do another point release of Sail to push that and various other smaller bug fixes before the next major release.

Even with that there are still several PMP related stubs you need before the Isabelle theory file will build so I will add those.

Incarnation-p-lee, 2024-04-04:
> This specific error is fixed in Sail by:
> 
> [rems-project/sail@297e4d2](https://github.com/rems-project/sail/commit/297e4d270acc6d273d27ad80cb3bf0ef78f7f985)
> 
> I should probably do another point release of Sail to push that and various other smaller bug fixes before the next major release.
> 
> Even with that there are still several PMP related stubs you need before the Isabelle theory file will build so I will add those.

Great, will have a try for the next release and keep you posted.

jordancarlin, 2024-05-14:
Now that #444 and #464 are merged, can this be closed?

Mudassir10X, 2024-05-23:
I think this issue should be closed @Ludini1.