riscv/sail-riscv #401: Optimization

M20M19, 2024-02-05:
Hi, I am currently working on the following project: optimizing the Sail emulator for RISC-V. Couldn't you suggest what could theoretically be changed to speed up the execution of programs, what possible optimizations can be applied? Thanks in advance for your recommendations

jrtc27, 2024-02-05:
Where is this project coming from? Do you have a mentor you can ask these kinds of questions?

rsnikhil, 2024-02-06:
In next week's 'tech-golden-model' group meeting, there will be a presentation on exactly this topic by someone who has been working for some time on a JIT compiler for the Sail RISC-V model, with reportedly substantial performance improvement.

If you or your organization are members of RVI (RISC-V International), you can join the 'tech-golden-model' group and attend the meeting, and connect with the speaker.

M20M19, 2024-02-06:
> Where is this project coming from? Do you have a mentor you can ask these kinds of questions?

I need to clarify that this is my graduation project, and my mentor advised me to ask this question to the developers. For example, one of the students in our department has already defended his thesis on a similar topic. In this work, the author used low-level optimizations to speed up emulation, such as optimizations for string operations, as well as the use of alternative system call implementations in libc. I'm interested in what other alternative approaches can be used to speed up the Sail emulator.

Timmmm, 2024-05-07:
I've looked into this a fair bit. Here are some things that I found to speed it up, all found via profiling the C emulator using gperftools. All but the last are changes to the model code rather than Sail itself.

* Use a TLB size of 64 instead of 1 (only applies when using virtual memory)
* Replacing uses of `nat` or `int` with `range(...)` so that Sail can use `uint64_t` instead of GMP. I think the worst offender for this was in the virtual address translation code and is now gone.
* Some uses of `unsigned()` cause Sail to use GMP, e.g. the `within_` functions are pretty bad.
* Sail decides to use GMP at compile time, but it would be better if it switched between uint64_t and GMP at run-time because in most cases where it can't prove uint64_t is ok, in practice the values are never too big for it. I did start a patch for this but it's very tedious so I didn't finish it.

Timmmm, 2024-06-28:
TLB change is here: #498

Anyway I'll close this since it's a question and I think it has been answered: the main optimisation to do is switch between GMP and `uint64_t` at runtime depending on the actual values of `int`s or the lengths of bit vectors.

Btw there is also the Pydrofoil project which uses the Pypy JIT to run Sail code much faster than the current C output (I would guess mainly because it already has that optimisation due to Python integers being arbitrary precision).

closed 2024-06-28 by Timmmm