riscv/sail-riscv #409: Use vectors for registers

Timmmm, 2024-02-19:
As far as I can see there's no reason to have this:

```
register x1  : regtype
register x2  : regtype
register x3  : regtype
register x4  : regtype
register x5  : regtype
register x6  : regtype
...
```

Instead of

```
register xrf : vec(regtype, 32)
```

This is done for all the X, F and V registers.

Alasdair, 2024-02-19:
In the concurrency model, `x1` .. `xN` are separate registers for intra-instruction dependencies (so instructions modifying different registers can be re-ordered), whereas `xrf : vector(...)` would be a single register.

jrtc27, 2025-03-26:
Hm, that seems less than ideal. Should the concurrency model not differentiate between vector-of-registers and register-of-vectors?

Alasdair, 2025-03-26:
The concurrency model does differentiate between vector-of-registers and register-of-vectors, I think that's the issue here.

The issue with tracking dependencies at a sub-register granularity is we'd lose the ability to statically determine an instruction footprint (at least in an easy way). It also just seems like the easier thing to say that a single sail `register` represents the unit of atomicity for dependency tracking. Anything smaller we'd have to pick some point between that and tracking dependencies at the level of individual bits which would be impractical.

jrtc27, 2025-03-26:
Oh sorry, I glanced and saw `vector(regtype, ...)` and interpreted that as vector-of-register, but that's not true, it's `register ... : vector(regtype, ...)` so the whole thing is the register. My bad for not fully reading the original issue.