riscv/sail-riscv #410: CSRs should use scattered functions

Timmmm, 2024-02-19:
The CSR code is a bit odd. For example `readCSR()` uses a `match`:

```
function readCSR csr : csreg -> xlenbits = {
  let res : xlenbits =
  match (csr, sizeof(xlen)) {
    /* machine mode */
    (0xF11,  _) => zero_extend(mvendorid),
...
```

but *also* a scattered function for extensions.

```
...

    _           => /* check extensions */
                   match ext_read_CSR(csr) {
                     Some(res) => res,
                     None()    => { print_bits("unhandled read to CSR ", csr);
                                    zero_extend(0x0) }
                   }
```

This is silly. It should just work like the scattered functions for instructions - everything uses scattered functions, and there's a `riscv_csrs_begin.sail` and `riscv_csrs_end.sail`, and you add new CSRs just by inserting extra files.

defermelowie, 2024-02-23:
This is related to my [question](https://github.com/riscv/sail-riscv/issues/311) a while back. I agree that CSRs should use scattered functions but the question then is whether or not non-standard extensions should keep using the `ext_is_CSR_defined`, `ext_read_CSR` and `ext_write_CSR` hooks or use the scattered functions you propose here. The latter seems more consistent with the way instructions work.

Timmmm, 2024-02-24:
Yes I agree there should be a single scattered function; no `ext_...`.

PR #543 (2024-09-25): Make CSR definitions scattered
Resolve the strange split between `read_CSR` (which is not `scattered`) and `ext_read_CSR` (which is). Same for `write_CSR` and `is_CSR_defined`.

I changed the return type of `read_CSR` from `option(xlenbits)` to `xlenbits` since the code must already check `is_CSR_defined` before calling `read_CSR`. The only function that returned `None()` was the `seed` CSR in `write_seed_csr`, which actually meant you would get a weird "Unhandled write to CSR" if you wrote to `seed`.

I renamed & reordered the files slightly to make the scattered mapping work, but I haven't m [... cut]
commit b63b12f2 (2024-09-25): Make CSR definitions scattered

Resolve the strange split between `read_CSR` (which is not `scattered`) and `ext_read_CSR` (which is). Same for `write_CSR` and `is_CSR_defined`.

I changed the return type of `read_CSR` from `option(xlenbits)` to `xlenbits` since the code must already check `is_CSR_defined` before calling `read_CSR`. The only function that returned `None()` was the `seed` CSR in `write_seed_csr`, which actually meant you would get a weird "Unhandled write to CSR" if you wrote to `seed`.

I renamed & reordered the files slightly to make the scattered mapping work, but I haven't  [... cut]
closed 2024-09-25 by Timmmm