riscv/sail-riscv #412: Remove `instbits`

Timmmm, 2024-02-23:
Currently there's a global register:

```
/* internal state to hold instruction bits for faulting instructions */
register instbits : xlenbits
```

Which is saved after fetch:

```
                F_Base(w) => {
                  instbits = zero_extend(w);
```

And used in `handle_illegal()`:

```
function handle_illegal() -> unit = {
  let info = if plat_mtval_has_illegal_inst_bits ()
             then Some(instbits)
             else None();
```

This use of a global is unnecessary. It would be more elegant to:

1. Remove the `instbits` register.
2. Add `RETIRE_ILLEGAL` to the `Retire` enum.
3. Change all instances of `handle_illegal(); RETIRE_FAIL` to just `RETIRE_ILLEGAL`
4. Add `instbits` as a parameter to `handle_illegal()`.
5. Call `handle_illegal(...)` from `step()` if `retired == RETIRE_ILLEGAL`.

Timmmm, 2024-02-23:
Good call, that does sound better and I guess would let you treat other exceptions in the same way (I'll have to double check the code though). I don't think `exception_details` needs to include the instructions bits though since they are already available at the point in `step()` where it's handled (and they'd be a bit awkward to get in `execute()` since the instruction has already been decoded).

jrtc27, 2024-02-23:
> Good call, that does sound better and I guess would let you treat other exceptions in the same way (I'll have to double check the code though). I don't think `exception_details` needs to include the [...]

Sorry I should have been clearer: instbits would be supplied to handle_exception to use in the case that the execute clause returned Exc_Illegal_Inst (names entirely made up, any semblance to real names is coincidental). Or, to put it another way: yes, that's what I meant, I just failed to clearly communicate it.

martinberger, 2024-03-19:
I'm also in favour of extending the `Retired` data type. There is another important case: debug triggers. You want those to be handled uniformly in `step` as well. But if you get a trigger for example of a load address, that trigger is found in the `execute` clause of the relevant load instruction. For this purpose a `RETIRE_TRIGGER(...)` would be neat.

commit 49d1c8e5 (2025-04-16): Refactor instruction retirement.

The eventual goal here is to surface wait-states of the hart
(e.g. as caused by WFI and WRS.{NTO,STO}) into the stepper,
and handle debug triggers from execution for extensions such as
Sdtrig.

The refactor makes interrupt and exception handling more
symmetrical: both are now handled in the stepper.

This also removes the model-internal global instbits register
into a stepper-local variable, which fixes #412.

This refactor is based on ideas discussed in #398 and #412.
closed 2025-04-16 by pmundkur