riscv/sail-riscv #428: Syntax Error in 'riscv_sys_regs.sail' File

Sowmya3062, 2024-03-11:
I encountered a syntax error while attempting to clone this repository and run 'make'. The error message I received is as follows:

Syntax error:
[model/riscv_sys_regs.sail]:120:25-25
120 |    else [m with F = v[F], D = v[D] & v[F]]
    |                         ^
    | current token: ,
make: *** [Makefile:218: generated_definitions/ocaml/RV64/riscv.ml] Error 1

Alasdair, 2024-03-11:
What version of Sail do you have? That should be valid syntax for Sail >= 0.15

Note that I don't think just plain `make` usually works, as it tries to do everything and will fail for some target on most systems (but with a different error). You should pick the makefile target you actually want, like `make csim` for the C simulator and so on.

Sowmya3062, 2024-03-12:
The syntax is valid for version 0.17.1, thanks!

closed 2024-03-12 by Sowmya3062