riscv/sail-riscv #459: VM code reads 8 bytes even for 4-byte PTEs

Timmmm, 2024-04-26:
In `pt_walk`:

```
  // Read this-level PTE from mem
  let mem_result = mem_read_priv(Read(Data),              // AccessType
                                 Supervisor,              // Privilege
                                 pte_phys_addr,
                                 8,                       // atom (8)
                                 false,                   // aq
                                 false,                   // rl
                                 false);                  // res

  match mem_result {
    MemException(_) => PTW_Failure(PTW_Access(), ext_ptw),
    MemValue(pte)   => {
      let pte_flags = Mk_PTE_Flags(pte[7 .. 0]);
```

Should be:

```
  // Read this-level PTE from mem
  let mem_result = mem_read_priv(Read(Data),              // AccessType
                                 Supervisor,              // Privilege
                                 pte_phys_addr,
                                 2 ^ sv_params.log_pte_size_bytes,
                                 false,                   // aq
                                 false,                   // rl
                                 false);                  // res

  match mem_result {
    MemException(_) => PTW_Failure(PTW_Access(), ext_ptw),
    MemValue(pte)   => {
      // Extend to 64 bits even on RV32 for simplicity.
      let pte : bits(64) = zero_extend(pte);

      let pte_flags = Mk_PTE_Flags(pte[7 .. 0]);
```

Tbh I think we should probably try to fully use Sail's fancy type system in this new VM code rather than just expanding everything to 64 bits immediately. It's very useful for catching bugs like this.

rsnikhil, 2024-04-29:
>    Tbh I think we should probably try to fully use Sail's fancy type system in this new
>    VM code rather than just expanding everything to 64 bits immediately. It's very useful for catching bugs like this.

We're currently building for either RV32/Sv32 or RV64/Sv39/Sv48.  But I believe the spec permits supporting all the options simultaneously in a single implementation.  The do-everything-at-64-bits regime in the current vmem code was geared towards doing that in the future; I don't know how we can restrict this with types.

Timmmm, 2024-04-29:
> I don't know how we can restrict this with types.

You should be able to make the widths of everything type parameters. One issue is that there are quite a lot of type parameters. However you could in theory just have a single type parameter for the whole thing, based on the `Sv??` number - i.e. `{32, 39, 48, 57}`. Then you can use Sail's very recent support for `if/else` in types to derive all the other parameters where needed (I don't think this was an option when you wrote this code). There's an example of that [here](https://github.com/riscv/sail-riscv/pull/448/files#diff-338285f5f239a4e9e139ce54188bf2341eeb4040751cdbfdc982a8dcabd40a39R4):

```
val fsplit : forall 'n, 'n in {16, 32, 64}.
  bits('n) -> (
    bits(1),
    bits(if 'n == 16 then 5 else (if 'n == 32 then 8 else 11)),
    bits(if 'n == 16 then 10 else (if 'n == 32 then 23 else 52)),
  )
```

So for example instead of:

```
// PRIVATE: extract PPNs of PTE
function PPNs_of_PTE(sv_params : SV_Params, pte : bits(64)) -> bits(64) = {
  let mask : bits(64) = zero_extend(ones(sv_params.pte_PPNs_size_bits));
  (pte >> sv_params.pte_PPNs_lsb_index) & mask
}
```

You can condition it on the PTE size, so that it returns `bits(22)` or `bits(44)`.

```
val PPNs_of_PTE : forall 'pte_size, 'pte_size in {32, 64}.
  bits('pte_size) -> bits(if 'pte_size == 32 then 22 else 44)
function PPNs_of_PTE(pte) = if sizeof('pte_size) == 32 then pte[31 .. 10] else pte[53 .. 10]
```

It is a fair bit of work to make Sail happy though. I might have a go one day.

PR #461 (2024-05-20): Don't read or write 8 bytes for 4-byte PTEs
For Sv32 Page Table Entries are only 4 bytes, but the old code was unconditionally reading and writing 8 bytes.

Fixes #459
commit 4d607432 (2024-05-20): Don't read or write 8 bytes for 4-byte PTEs

For Sv32 Page Table Entries are only 4 bytes, but the old code was unconditionally reading and writing 8 bytes.

Fixes #459
closed 2024-05-20 by billmcspadden-riscv