riscv/sail-riscv #492: help with compiler error

ved-rivos, 2024-05-28:
I am seeing this error:

```
File "riscv.ml", line 9649, characters 40-56:
9649 |   let znum = (if (zbit_to_bool (access (zvalid_ctr_depth, Big_int.zero))) then (Big_int.of_int (16)) else Big_int.zero) in
                                               ^^^^^^^^^^^^^^^^
Error: This expression has type Sail_lib.Big_int.num = Z.t
       but an expression was expected of type 'weak9 list
Command exited with code 2.
Compilation unsuccessful after building 15 targets (14 cached) in 00:00:00.
make: *** [Makefile:234: ocaml_emulator/_sbuild/riscv_ocaml_sim.native] Error 10

```
I believe its complaining about something in this code:

```
platform.ml:let config_valid_ctr_depth = ref Big_int.zero
platform.ml:let set_valid_ctr_depth x   = config_valid_ctr_depth := Big_int.of_int x
platform.ml:let valid_ctr_depth ()          = !config_valid_ctr_depth
riscv_ocaml_sim.ml:                           
                              ("-valid-ctr-depth",
                           Arg.Int P.set_config_valid_ctr_depth,
                           " valid CTR depths");
```

jrtc27, 2024-05-28:
What's the Sail side of valid_ctr_depth?

ved-rivos, 2024-05-28:
`val sys_valid_ctr_depth = {c: "sys_valid_ctr_depth", ocaml: "Platform.valid_ctr_depth", _:                    "sys_valid_ctr_depth"} : unit -> bits(64)`

jrtc27, 2024-05-28:
Big_int is for mathematical integer types, not bit vectors. See arch_bits_of_int to convert from an int (not big) to that, as used by things like dram_size.

ved-rivos, 2024-05-28:
Thanks. That typecast worked.
```
platform.ml:let config_valid_ctr_depth = ref Int.zero
platform.ml:let set_valid_ctr_depth x = config_valid_ctr_depth := x
platform.ml:let valid_ctr_depth () = arch_bits_of_int !config_valid_ctr_depth
riscv_ocaml_sim.ml: Arg.Int P.set_valid_ctr_depth,

```

closed 2024-05-28 by ved-rivos