riscv/sail-riscv #503: How to generate a reference ISA model in SystemVerilog?

SeddonShen, 2024-06-29:
Can this project generating a reference ISA model in SystemVerilog now?
In the Readme it says that this work is in progress, are there any milestones?
Thanks!

Timmmm, 2024-06-29:
There's some support but it's incomplete. For example loops aren't implemented. I believe it has been used for some formal equivalence checking but I'm not sure of the details.

Alasdair, 2024-06-29:
It's available on the current development branch of Sail. It's in a bit of a transitional state right now as we implement support for generating systemverilog using modules as an alternative to a more procedural style, as well as refactoring the translator to improve its capabilities.

A previous version prior to this refactoring was able to generate a complete translation of this model (prior to the addition of the vector extension, which complicates things greatly). We also had a complete CHERI translation we used for verification. Hopefully soon I will be able to polish the new implementation and get it tested on the current model (But probably not before the POPL deadline this month...). Loops were handled by unrolling.

closed 2024-07-01 by SeddonShen