riscv/sail-riscv #507: Many compilation warnings in OCaml emulator: ignored-partial-application

Kakadu, 2024-07-03:
The CI version builds OCaml emulator with a lot of warnings. The most severe of them is 
````
File "riscv.ml", line 12439, characters 4-48:
12439 |     Softfloat.f64_round_to_int (zrm, zv, zexact);
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Warning 5 [ignored-partial-application]: this function application is partial,
maybe some arguments are missing.
````
The compiler says that this function is not really called, and extra argument(s) seems to be forgotten. This means that correctness of the emulator could be in danger.

Timmmm, 2024-07-04:
Yeah I noticed this too. The OCaml emulator doesn't actually support float currently anyway so it's probably not a big deal. This will change when the float implementation is moved from Berkeley softfloat to native Sail, which which case these functions will be deleted anyway.

jordancarlin, 2024-10-12:
This can be closed now that the OCaml emulator is gone

closed 2024-10-12 by Alasdair