riscv/sail-riscv #509: Remove OCaml emulator

Timmmm, 2024-07-04:
Can we remove the OCaml emulator? I think this would have significant benefits:

* Less confusing for new users - they no longer need to choose between two very similar but different emulators.
* We no longer have to implement emulator features twice.
* Contributors don't need to learn OCaml.
* Shorter CI time.
* When we use the binary release of the Sail compiler, this will mean we can entirely drop OCaml as a dependency, even in CI.

trdthg, 2024-07-05:
Discussions from the past: #138

martinberger, 2024-07-08:
Another problem with the Ocaml emu is 

- Very slow!

arichardson, 2024-07-09:
I've also found having to add the ocaml callbacks for every new features a bit annoying in the past (especially since I can't write ocaml and just copy-paste). So +1 from me.

jordancarlin, 2024-08-12:
Is this something that we're planning to proceed with? I can look into it if so.

Timmmm, 2024-08-12:
Yeah in the weekly meeting we basically agreed to do it and I said I would make a PR, but I've only ~30% completed it.

One thing we didn't discuss which is a unique OCaml backend feature is the thing where it can generate random instructions based on the `encdec` mapping, which I believe is used by TestRig. It doesn't depend on the emulator or OCaml callbacks but we might want to still keep the "generate OCaml code" Makefile targets which it presumably uses (I haven't checked yet).

PR #544 (2024-09-19): Remove the OCaml emulator
This has the following benefits:

* Reduces the maintenance burden for making changes to the emulator. They don't have to be done in two places and you don't need to learn OCaml.
* Removes the dependency on OCaml and OPAM, which is a weirdly bug-prone tool.
* Removes confusion for new users about which emulator to use.
* Shorter CI time.

<strike>Note, I left the targets to actually build the OCaml code since I believe it is used by TestRig for its random instruction generation.</strike>

Fixes #509
commit bdf95b07 (2024-09-19): Remove the OCaml emulator

This has the following benefits:

* Reduces the maintenance burden for making changes to the emulator. They don't have to be done in two places and you don't need to learn OCaml.
* Removes the dependency on OCaml and OPAM, which is a weirdly bug-prone tool.
* Removes confusion for new users about which emulator to use.
* Shorter CI time.

It removed two capabilities which were deemed to be acceptable for now:

* Unlike the OCaml emulator the C emulator cannot dynamically generate a Device Tree Blob based on the options you give it. However this would be relatively ea [... cut]
closed 2024-09-19 by Timmmm