riscv/sail-riscv #514: make riscv_coq_build fails with a type error

yoichi-nexus, 2024-07-15:
I am looking for a specification language for an architecture that's very similar to RISC-V. So I was trying the instructions in the README.

I installed `sail` from `opam`
```
$ sail -v
Sail 0.17.1 (sail @ opam-v2.2.0)
```

When I typed `make riscv_coq_build` I saw an error:

```
Type error:
model/riscv_termination_rv64.sail:9.20-26:
9 |termination_measure walk39(_,_,_,_,_,_,level,_, _) = level
  |                    ^----^
  | No type declaration found for walk39
```

This might be because I could not follow
```
When building these targets, please make sure the corresponding prover libraries in the Sail directory ($SAIL_DIR/lib/$prover) are up-to-date and built, e.g. by running make in those directories.
```
even when I tried with a `SAIL_DIR` variable pointing at a checkout. That was because the [sail source tree](https://github.com/rems-project/sail/tree/d8c5322fbde0049d844eb1f2e2cc8c56498e1acc/lib) does not have `lib/coq` in it.

Timmmm, 2024-07-15:
Ah this is actually because those functions are from a previous implementation of the virtual memory system. The `termination_measure` files are only used for formal verification and that happens on an ad-hoc basis so it looks like these files weren't updated properly.

You could try just deleting their contents. I guess depending on how far you get into Coq you might need to add a new `termination_measure` for the new page table walk function, something like this:

```
termination_measure pt_walk(_,_,_,_,_,_,_,level,_,_) = level
```

Or you could go to the commit just before the one that introduced `pt_walk`. It wasn't too long ago.

bacam, 2024-07-15:
Yes, unfortunately the Coq support has bitrotted a little.  I intend to provide a pull request to update it after a new Sail release, which should be quite soon.  For now, I suggest looking at the branch in pull request #386, which is against an older version of the model but should work.  Note that it might still require building a newer version of Sail than the last release, however.

yoichi-nexus, 2024-07-15:
Now I'm sensing that efforts into solving this one would be superseded by @bacam 's larger efforts. So I am closing this issue. Thank you very much for the quick responses.

(Context: I think I'm going to define something in sail. Initially I'll probably find the SMT output very useful. After some months interactive theorem provers become more relevant. This is all quick speculation.)

closed 2024-07-15 by yoichi-nexus

miguelis, 2024-12-30:
@bacam Hey. I’ve seen that you’ve created a more updated [branch](https://github.com/bacam/sail-riscv/tree/coq-update-202409). I’d like to know if this branch is up to date with the current version of RISCV or if some features are missing. Thank you very much for your help.

bacam, 2025-01-06:
It's about 6 months behind the current model, but hopefully won't be too hard to rebase.  You would also need a suitable version of Sail, which is currently the `coq-new-regstate` branch until I fix an issue caused by improvements to Sail's type checking.  Hopefully I'll be able to sort both of these out shortly...