riscv/sail-riscv #529: The sideleg CSR might be deleted

KotorinMinami, 2024-08-27:
Recently, I reviewed the latest specification and found that the "sideleg" register no longer exists in the CSR table. Upon further investigation, it appears to have been removed. However, in the sail model, this register and its encoding still exist. Should this be removed to avoid potential issues with illegal encodings in the sail?

jrtc27, 2024-08-27:
It's gated on Ext_N, which there isn't a way to enable unless you patch the source. The only issue that would arise is if the encoding is reused for another CSR. Though it would probably be wise to remove the implementation given the obsolete draft was removed outright from the riscv-isa-manual repo.

jordancarlin, 2024-10-16:
Closed by #584