riscv/sail-riscv #555: Check that read_vmask_carry is correct

Timmmm, 2024-09-20:
> One thing I noticed is the read_vmask_carry function doesn't seem to be doing anything, as it looks like it is only called with vm=0b0 where it is then the same as read_vmask.

From [here](https://github.com/riscv/sail-riscv/pull/552#issue-2537766943).

KotorinMinami, 2024-10-15:
After looking though the code, I found that the read_vmask function and the read_vmask_carry function are different.In the read_vmask function, result is initialized to vector_init(true), while in the read_vmask_carry function, it is initialized to vector_init(false).
```
val read_vmask : forall 'n, 'n >= 0. (int('n), bits(1), regidx) -> vector('n, bool)
function read_vmask(num_elem, vm, vrid) = {
  assert(num_elem <= sizeof(vlenmax));
  let vreg_val : vregtype = V(vrid);
  var result   : vector('n, bool) = vector_init(true);           //here
...

val read_vmask_carry : forall 'n, 'n >= 0. (int('n), bits(1), regidx) -> vector('n, bool)
function read_vmask_carry(num_elem, vm, vrid) = {
  assert(num_elem <= sizeof(vlenmax));
  let vreg_val : vregtype = V(vrid);
  var result   : vector('n, bool) = vector_init(false);        //and here
```
With the initialization different, we can see that the results of the read_vmask and read_vmask_carry functions have different uses. One is to record whether the element is operated(the default is all selection), and the other is to record whether the element has a carry from the previous element(and the default is none).

Timmmm, 2025-02-25:
Ah yes you're right. I think the code could be simplified but it doesn't seem wrong at least!