riscv/sail-riscv #576: Outdated Status.md file

jordancarlin, 2024-10-05:
The [docs/Status.md](https://github.com/riscv/sail-riscv/blob/master/doc/Status.md) has fallen very far out of date. It is missing many of the extensions now included in the model and only includes a small subset of the supported configuration options. 

It should either be updated or removed. I'd be in favor of removing it since it clearly hasn't been missed, but could also see a case for wanting to keep it and ensure that it stays up to date going forward.

trdthg, 2024-10-18:
How about moving it to the README? For example, [Spike](https://github.com/riscv-software-src/riscv-isa-sim#about) declares the extensions and version it currently supports in the README.

jordancarlin, 2024-10-18:
Makes sense to me

PR #602 (2024-10-30): Documentation Updates: reorder README and combine with STATUS.md
Closes #576, the current `STATUS.md` files has fallen very far out of date. This removes it in favor of a list of supported extensions/features in the main README. I also reordered the README to put the "Getting Started" section near the top and did some other general cleanup with it.

Also wondering if it is worth keeping the entire section on what Sail is. We already link to the Sail repo, so it might make more sense to remove this and just include a quick sentence or two overview.

Going through the list of supported extension [... cut]
closed 2024-10-30 by Timmmm