riscv/sail-riscv #579: Confusing assembly clause : VSUXSEGTYPE

moste00, 2024-10-06:
As seen in the screenshot below, there are 2 assembly clauses to stringify the AST case `VSUXSEGTYPE`, they are exactly identical except for a single-character change in mnemonic of the instruction.

I wanted to ask:

(1) The behaviour of this clause is that AST->string conversion will always pick the first version, but string->AST conversion will accept both mnemonics and return the same AST case. Is that correct? 

(2) Is the behaviour in (1) intended?

(3) Assuming (2) to be "yes", is there a less confusing way of phrasing this logic? Perhaps by using a helper function or mapping to consume both the `vsox` and `vsux` mnemonics in one pass?

Permalink to the source file https://github.dev/riscv/sail-riscv/blob/b90ec7881eb0a0ecea6521e3d6cc03bf0b057e41/model/riscv_insts_vext_mem.sail#L626.

[image]

Timmmm, 2024-10-08:
It's a typo - see #577

moste00, 2024-10-11:
@Timmmm thanks, closed.

closed 2024-10-11 by moste00