riscv/sail-riscv #583: Merge 'with' updates

Timmmm, 2024-10-09:
There's quite a lot of code like this:

```
function lower_mstatus(m : Mstatus) -> Sstatus = {
  let s = Mk_Sstatus(zero_extend(0b0));
  let s = [s with SD = m[SD]];
  let s = set_sstatus_UXL(s, get_mstatus_UXL(m));
  let s = [s with MXR = m[MXR]];
  let s = [s with SUM = m[SUM]];
  let s = [s with XS = m[XS]];
  let s = [s with FS = m[FS]];
  let s = [s with VS = m[VS]];
  let s = [s with SPP = m[SPP]];
  let s = [s with SPIE = m[SPIE]];
  let s = [s with UPIE = m[UPIE]];
  let s = [s with SIE = m[SIE]];
  let s = [s with UIE = m[UIE]];
  s
}
```

which should be written like this:

```
function lower_mstatus(m : Mstatus) -> Sstatus = {
  let s = Mk_Sstatus(zero_extend(0b0));
  [s with
    SD = m[SD],
...
    SPP = m[SPP],
    SPIE = m[SPIE],
    UPIE = m[UPIE],
    SIE = m[SIE],
    UIE = m[UIE],
  ]
}
```

jordancarlin, 2024-10-15:
Can be closed with #586 merged

Timmmm, 2024-10-16:
I think there's a few more that could be merged:

* `legalize_mstatus`
* `lower_mstatus` (mostly done but we missed `SD`)
* `lift_sie`

closed 2024-11-06 by Timmmm