riscv/sail-riscv #587: NA4 Region not calculated correctly

MuhammadHammad001, 2024-10-11:
The implementation for address upper range calculation for NA4 is not correct. Please refer to:
https://github.com/riscv/sail-riscv/blob/f7192a642242a718290524a77797b11fcf6783b6/model/riscv_pmp_control.sail#L18C1-L23C16 
```
    NA4   => {
               // NA4 is not selectable when the PMP grain G >= 1. See pmpWriteCfg().
               assert(sys_pmp_grain() < 1, "NA4 cannot be selected when PMP grain G >= 1.");
               let lo = pmpaddr;
               Some((lo, lo + 4))
             },
```
The correct implementation should be:
```
    NA4   => {
               // NA4 is not selectable when the PMP grain G >= 1. See pmpWriteCfg().
               assert(sys_pmp_grain() < 1, "NA4 cannot be selected when PMP grain G >= 1.");
               let lo = pmpaddr;
               Some((lo, lo + 1))
             },
```
This will properly configure the NA4 region equal to 4B instead of 16B.

MuhammadHammad001, 2024-10-11:
PR # [588](https://github.com/riscv/sail-riscv/pull/588#issue-2580602424)

Timmmm, 2024-10-11:
Ah good spot. This got missed when we updated the function to return words instead of bytes (to fix 34 bit physical addresses on RV32). How did you find this bug out of interest? (I think our internal testing didn't find it because our CPU doesn't support NA4 :-/ ).

MuhammadHammad001, 2024-10-11:
I am actually writing the tests for PMP and SV-32 Virtual Memory and the
region was being set for 16B instead of 4B, therefore causing the tests for
NA4 to fail.

closed 2024-10-11 by Timmmm

allenjbaum, 2024-10-11:
Good job Hammad!