riscv/sail-riscv #614: Improve `to_bits` type checking

Timmmm, 2024-11-11:
Currently `to_bits` just silently accepts larger integers than will fit in the bitvector and slices off the top. That's a bit sad from a type checking point of view. Sail can perfectly happily check this at compile time:

```
val to_bits : forall 'l 'x, 'l >= 0 & 0 <= 'x < 2 ^ 'l . (int('l), int('x)) -> bits('l)
function to_bits (l, n) = get_slice_int(l, n, 0)
```

I think we should change the definition to that. There are some places where the slicing is intentional (e.g. the bit rotate functions), where we can probably introduce an explicit function:

```
val to_bits_truncate : forall 'l, 'l >= 0 . (int('l), int) -> bits('l)
function to_bits_truncate (l, n) = get_slice_int(l, n, 0)
```

And there are a lot of places where it probably isn't intentional, where we could add

```
val to_bits_unsafe : forall 'l, 'l >= 0 . (int('l), int) -> bits('l)
function to_bits_unsafe (l, n) = get_slice_int(l, n, 0)
```

with TODOs to add the appropriate assertions & types. A lot of them are in the vector extension which has very loose typing so it will take a while to fix them there.

jordancarlin, 2024-11-11:
Seems like a great type safety improvement that makes a lot of sense.

Timmmm, 2025-05-21:
This is done now except for #960 and all of the vector code, which generally needs better types before they are precise enough for the new `to_bits()`.

Timmmm, 2025-07-31:
Finished except for all the `to_bits_unsafe()` in the vector code.