riscv/sail-riscv #632: Support for mstatus MBE, SBE

MuhammadHammad001, 2024-12-02:
Hi
I am currently writing the test for little and big endianess in the virtual memory verification.
I was trying to write the mstatush.SBE to enable big endianess for loads/stores but I am unable to do so.
```
mem[X,0x80000A98] -> 0xA073
mem[X,0x80000A9A] -> 0x3102
[643] [M]: 0x80000A98 (0x3102A073) csrrs zero, 0x310, t0
CSR 0x310 -> 0x00000000
CSR 0x310 <- 0x00000000 (input: 0x00000010)
```
I went through the sail code to figure out what is wrong and I saw the following comment [here](https://github.com/riscv/sail-riscv/blob/a365bd24cd1905408798b21f16718e0481683146/model/riscv_sys_regs.sail#L275C1-L279C2):
```
  /* We don't currently support changing MBE and SBE. */
  if xlen == 64 then {
    Mk_Mstatus([m.bits with 37 .. 36 = 0b00])
  } else m
}
```

Is there any plan to add the support for this or am I missing something?

jordancarlin, 2024-12-03:
The [OpenHWGroup CVW core](https://github.com/openhwgroup/cvw) implements support for switching endian modes using MBE/SBE. Would be a great feature to get into the Sail model eventually, but I don't think it's been that high on anyone's priority list.

Timmmm, 2024-12-03:
Has anyone actually taped it out with support for big endian though? Kind of feels like CVW is trying to support every feature of RISC-V regardless of whether anyone wants it (which is fine), and I would be mildly surprised if anyone really wants big endian! I guess someone asked for it though; the spec says it was added to support legacy code.

Anyway feel free to try. It might not be too invasive; you probably just need to modify `mem_read_priv_meta` and `mem_write_value_priv_meta` to swap the bytes (unless `Typ` is `Execute`).

It may interact awkwardly with #49 - maybe have a skim of Alasdair's PR there before doing anything.

MuhammadHammad001, 2024-12-03:
@jordancarlin Sure, I can work on this

jordancarlin, 2025-02-12:
@MuhammadHammad001 Is this something you ended up working on/are planning to work on? If not I might take a stab at it soon after all.

MuhammadHammad001, 2025-02-12:
Hi Jordan,
I actually completed this work a while ago but could not find the time to
send a PR. I will send it today.

jordancarlin, 2025-02-13:
> I actually completed this work a while ago but could not find the time to
> send a PR. I will send it today.

Fantastic!