riscv/sail-riscv #633: Make `extension` enum not scattered

Timmmm, 2024-12-05:
Currently we have a scattered `extension` enum that we generally add entries to in the files that implement the extension, e.g. `riscv_svinval.h` contains:

```
enum clause extension = Ext_Svinval
function clause extensionEnabled(Ext_Svinval) = sys_enable_svinval()
```

I've found that this causes quite annoying circular dependencies, especially with functions that depend on lots of extensions (e.g. stuff around legalizing `mie`, `mstatus`, `misa`, etc.).

I propose making `extension` not scattered, and just putting it all in a big list in one file (`riscv_extensions.sail`)

```
enum extension = {
  Ext_S,
  Ext_Svinval,
  Ext_...
}
```

This can then be added early on in the compile order. I don't think there are any real downsides to this and it might actually be quite nice to see them all in one place.

The `extensionEnabled()` clauses should still be scattered.

jordancarlin, 2024-12-05:
The only downside that I see is in relation to the new module system that @Alasdair is working on in #572. This would require all of the extensions to be defined in the enum even when that extension isn't included. I don't think that is a significant issue though because the `extensionEnabled` clause can still be in the extension specific files. Overall seems worthwhile solely for sake of dependencies.

Timmmm, 2024-12-05:
Yeah I think that's fine. If Sail ever has unused code warnings it might be annoying but I guess we can just add something like Rust's `#[allow(unused)]`.

PR #634 (2024-12-30): Move extension scattered enum to single file
Having this distributed through the code can lead to slightly awkward dependency order issues. We need to keep it scattered though so e.g. CHERI can add its extensions.

Fixes #633
closed 2024-12-30 by Timmmm