riscv/sail-riscv #640: sail-riscv Model May Enters Infinite Loop in RISC-V Arch Test PMP Cases

noionion, 2024-12-11:
The sail-riscv model seems to enter an infinite loop in the following pmp cases of riscv-arch-test. The specific sail configuration used is as follows: `--pmp-count=16 --pmp-grain=12`. I am not sure whether the issue lies with sail, my configuration, or the riscv-arch-test pmp cases themselves.

[image]
[image]

(the DUT is our core ip)

allenjbaum, 2024-12-11:
> This is (as of last night) a known issue. Some  tests were not designed to
> handle the case where pmp grain != 0 (especially the NA4 test, which
> shouldn't even be run with pmpgrain !=0)

Specifically - this likely is a test issue. not a Sail issue.

noionion, 2024-12-12:
Thank you for your reply! This issue will be closed.

closed 2024-12-12 by noionion