riscv/sail-riscv #653: bfloat16 support?

wanghuibin0, 2024-12-27:
Is there any plan to support bfloat16 related extensions?
Now that float16 in sail-riscv is represented as bits(16), how will be bfloat16 represented, if I model those bf16 involved instructions?

Timmmm, 2024-12-28:
I think it's been mentioned in the past but nobody is working in it currently. At the moment all the floating point stuff is implemented with the C Berkley softfloat library but we will eventually move it to [a pure Sail implementation](https://github.com/rems-project/sail/tree/sail2/lib/float) that @Incarnation-p-lee  has been working on so probably he knows best.

I would imagine it would be separate from the generic code, and use a `newtype` of `bits(16)`.

Incarnation-p-lee, 2024-12-30:
The pure sail implementation doesn't cover the BFloat format for now, but do we have something like BF32, BF64, and BF128?
For BF16, there will be newtype I think.

jordancarlin, 2024-12-30:
> The pure sail implementation doesn't cover the BFloat format for now, but do we have something like BF32, BF64, and BF128?

As far as I know there is only a 16 bit BFloat format.

Timmmm, 2024-12-30:
@Incarnation-p-lee bfloat16 is a non-IEEE floating point format, different to the IEEE binary16 format. IIRC it's just the IEEE binary32 format truncated to 16 bits, so it has way too many exponent bits, but conversion to/from normal single precision float is trivial. Popular in AI, at least it was a few years ago. I think hardware IEEE half support (Zfh) is more common now so I'm not sure how relevant it is today.

Incarnation-p-lee, 2024-12-31:
Yes, AFAIK, BF16 is mostly popular in the AI domain. I bet it comes from the [riscv bf16 spec](https://github.com/riscv/riscv-bfloat16/blob/main/doc/riscv-bfloat16.adoc).
However, given the current pure float implementation works on bits, maybe we can reuse most of exiting code and reconcile the types in the first insights.

BTW, I think both llvm and GCC support `zfh` and `zfb` already.

aamartin0000, 2025-01-09:
As far as I know, there is a notion of adding full BF16 arithmetic to the vector ISA, but not to scalar FP.  The BF16 arithmetic is mainly for AI/ML, and maybe mostly to support IME (the Integrated Matrix Extension). Also for AI/ML, there are other narrow FP formats under consideration (OCP FP8/6/4, OCP MX8/6/4) but no strong advocates.

I haven't looked in a while, but it should be fairly easy to implement BF16 in SoftFloat, and port to Sail if not done directly.

The (fairly) trivial conversions BF16 to/from FP32 for both scalar and vector are ratified, see Zfbf16 and similar.