riscv/sail-riscv #656: Locked bit not set in M Mode but still access fault on boundary crossing access

MuhammadHammad001, 2024-12-31:
Hi @Timmmm ,
@UmerShahidengr and I were writing tests for Physical Memory Protection and we came across an issue with the RISC-V Sail Model. As the spec says about the accesses which are made on the boundary of two regions with PMP enabled (S, U mode or in M Mode with Lock bit enabled).
[image]

Therefore, when we set the PMP NA4 region and we try to perform load/store double word in S/U mode with pmp configured we get load/store fault respectively as expected. But, in M-mode even when the lock bit is disabled, we get those faults which is wrong. The following [part](https://github.com/riscv/sail-riscv/blob/b92a3fa82e99435797c65c3a6f5d6c6688369f3c/model/riscv_pmp_control.sail#L92C1-L102C2
) of the RISC-V Sail Model is causing the problem.
```
function pmpMatchEntry(addr: physaddr, width: xlenbits, acc: AccessType(ext_access_type), priv: Privilege,
                       ent: Pmpcfg_ent, pmpaddr: xlenbits, prev_pmpaddr: xlenbits) -> pmpMatch = {
  let rng = pmpAddrRange(ent, pmpaddr, prev_pmpaddr);
  match pmpMatchAddr(addr, width, rng) {
    PMP_NoMatch      => PMP_Continue,
    PMP_PartialMatch => PMP_Fail,
    PMP_Match        => if   pmpCheckPerms(ent, acc, priv)
                        then PMP_Success
                        else PMP_Fail
  }
}
```
[image]

Here, you can see that even the Lock bit is disabled but we get load access fault.

I have corrected the implementation and now this load/store is successful.
[image]

Timmmm, 2024-12-31:
I don't understand the issue here. It says

> The matching PMP entry must match all bytes of an access, or the access fails, irrespective of the L, R, W, and X bits.

The partial match case must always fail, which is what the existing code implements.

MuhammadHammad001, 2024-12-31:
> The matching PMP entry must match all bytes of an access, or the access fails, irrespective of the L, R, W, and X bits.
Yep, it should only match when the PMP is active which is either in S/U mode or in M mode with Locked enabled.

> The partial match case must always fail, which is what the existing code implements.
The existing code:
```
function pmpMatchEntry(addr: physaddr, width: xlenbits, acc: AccessType(ext_access_type), priv: Privilege,
                       ent: Pmpcfg_ent, pmpaddr: xlenbits, prev_pmpaddr: xlenbits) -> pmpMatch = {
  let rng = pmpAddrRange(ent, pmpaddr, prev_pmpaddr);
  match pmpMatchAddr(addr, width, rng) {
    PMP_NoMatch      => PMP_Continue,
    PMP_PartialMatch => PMP_Fail,
    PMP_Match        => if   pmpCheckPerms(ent, acc, priv)
                        then PMP_Success
                        else PMP_Fail
  }
}
```
first calculates the address matching before even seeing whether the Locked bit is set in Machine mode or not which causes even in the cases when Lock bit is not set, the load/stores on the boundaries to fail without the PMP enabled.
Therefore, I updated the code to throw a fault only when the Lock bit is enabled, else the PMP is not active and the ld/sd should be successful even if we are accessing on the boundary of two regions.

MuhammadHammad001, 2024-12-31:
Yes, thank you for the clarification. It was a mistake on my end. :)

closed 2024-12-31 by MuhammadHammad001

allenjbaum, 2025-01-01:
It is a bit more nuanced than that, I thought.
 A misaligned load/strre can be broken up into two separate "accesses", so
if the first half is completely contained by one PMP entry,
and the second half by another, I thought the could be checked
independently; this is very implementation dependent
 (and happens even without PMPs, e.g. with MMU and 2 separate sets of page
table permissions)